facebookresearch/habitat-lab #562: Implement DAgger baseline

mathfac, 2021-01-14:
## 🚀 Feature

Implement DAgger baseline same way as it was added in https://github.com/facebookresearch/habitat-lab/issues/31. That would be great to add DAgger baseline to the arsenal of Embodied AI baselines in Habitat Lab.

## Motivation
DAgger is popular baseline for the task where optimization problem is hard to solve Implement and supervision required to train viable policy. DAgger approach can be implemented similarly how it was done in https://github.com/jacobkrantz/VLN-CE/tree/master/habitat_extensions as separate trainer but with usage of existing Encoders, Policies.

mathfac, 2021-01-14:
Tentative assignee is @jacobkrantz.

gchhablani, 2023-06-05:
Hi @mathfac
Is this issue still open?