facebookresearch/habitat-lab #566: Documentation for R2R/RxR Vision-and-Language Navigation

siddk, 2021-01-16:
## ❓ Questions and Help
Hey folks! 

I've been following the work on Habitat for a while, and think it is an incredible platform. I'm spinning up a few research projects here at Stanford that require using the Room2Room (and offshoots like R4R, RxR) datasets, and the graph-based navigation (used in papers like [this](https://arxiv.org/abs/1806.02724)).

I'm wondering if there's any documentation/tutorials for getting started with R2R in Habitat-Lab, and additionally, if you could point me to any VLN baselines that exist built on top of the Habitat ecosystem.

Thanks so much!

Skylion007, 2021-01-16:
Hey @siddk, long time to no see. :)

There are some baselines/examples that use R2R for VLN that have tests that can be found [here](https://github.com/facebookresearch/habitat-lab/blob/7766bf5fdb70db2aede35d2eb1123662a485f7f8/test/test_r2r_vln.py). It's pretty easy to add dataloaders for other variants like R2R. 

Off the top of my head, @jacobkrantz has a nice [VLN code baseline here](https://github.com/jacobkrantz/VLN-CE).

siddk, 2021-01-16:
Haha hey @Skylion007 - it's been a while! 

Thanks for the pointer, I'll check out the example.

The VLN-CE codebase by @jacobkrantz is for continuous environments - rather than navigating a set of discrete nodes in a MP3D environment, the space is continuous. For our work (comparing against existing baselines), we need the discrete environment; is this supported in Habitat-Lab (it isn't clear from [this example](https://github.com/facebookresearch/habitat-lab/blob/7766bf5fdb70db2aede35d2eb1123662a485f7f8/test/test_r2r_vln.py))?

Thanks!

Skylion007, 2021-01-16:
Yeah, actually discrete environments are the default. Continuous environments are relatively new (and somewhat experimental API). Should still be a lot of code in that repo that is relevant though.

Here is the config for the discrete VLN task baseline: https://github.com/facebookresearch/habitat-lab/blob/7766bf5fdb70db2aede35d2eb1123662a485f7f8/configs/tasks/vln_r2r.yaml

Skylion007, 2021-01-16:
Navigating to various discrete nodes can be easily done be passing the proper action_args to the TeleportAction define in the repo.

jacobkrantz, 2021-01-16:
>discrete environments are the default.

Actually continuous environments are the default in VLN-CE. @Skylion007 I think you are confusing discrete space with discrete actions. The VLN-CE task is continuous space but discrete actions.

@siddk I don't know of anyone doing straight VLN via node-to-node navigation in Habitat yet, but it is possible to implement via TeleportAction. You would also need to port the connectivity graph of R2R to Habitat which could be accomplished by mapping the location coordinates of each node from Matterport3D-Sim to Habitat. Not all of those locations are immediately navigable by a ground agent in continuous space, but depending on your use case you may or may not care. I'm happy to follow up further on this.

siddk, 2021-01-16:
Ahh gotcha - thanks for saving me @jacobkrantz! 

Node-to-Node navigation is ideally what I'd want to compare against some specific baselines (although we're definitely planning on moving to more of a VLN-CE setup at some point!).

Am I correct in assuming the best way to do this at the moment is through Peter Anderson's original Simulator ([via the Docker instructions here](https://github.com/peteanderson80/Matterport3DSimulator#installation--build-instructions))? Or is there something better?

It's interesting that this hasn't been folded into Habitat-Lab yet; I feel like a lot of recent work is still working on node-to-node navigation (R4R, RxR and affiliated models).

jacobkrantz, 2021-01-16:
If you want to make direct comparisons against specific VLN baselines, then sticking with the original VLN R2R task would be an easier place to start. Most works have used https://github.com/peteanderson80/Matterport3DSimulator, but you could also look into https://github.com/google-research/valan. If you wanted to do this in Habitat, recreating VLN would deviate from the VLN R2R task. Chiefly, there is a discrepancy of using 3D scene reconstructions (Habitat) vs using a graph topology of Matterport camera scans (VLN R2R task definition). Even if you match camera settings, there is some loss in visual fidelity from using reconstructions. If you then wanted direct comparison to VLN baselines, you would need to re-run them in your new Habitat setting.

This is just guiding you in the direction of least resistance -- I think it would be interesting to see node-to-node VLN in Habitat!

siddk, 2021-01-18:
Thanks so much @jacobkrantz! Really appreciate it. In the meantime, I'll proceed with the existing Simulator, and maybe re-implement the VALAN wrapper.

closed 2021-01-18 by siddk

Skylion007, 2021-01-18:
@siddk If you do reimplement the wrapper, feel free to submit a PR.