facebookresearch/habitat-lab #570: Reproduce the results of the DDPPO paper

rosanom, 2021-01-22:
## ❓ Questions and Help
Hi! First of all thank you for the amazing project you're carrying out!

I'm trying to reproduce the results obtained in the ddppo paper. I just installed the latest versions of habitat-sim and habitat-api, downloaded the pre-trained models, downloaded the "Gibson dataset for Habitat" (`Gibson_dataset_trainval`) and the corresponding Gibson task dataset from [here](https://github.com/facebookresearch/habitat-lab/blob/master/README.md#Gibson) (`pointnav_gibson_v1.zip` file).
Then I slighly modified the `habitat_baselines/config/pointnav/ddppo_pointnav.yaml` config file, to use the correct sensor (RGB or Depth) and load the correct pretrained checkpoint. If I run the `habitat_baselines/rl/ddppo/single_node.sh` with the `eval` flag the process freeze, I don't know the exact reason, maybe the procedure expects a checkpoint including the config parameters, that it isn't finding. For this reason I launched the training process from few seconds, in order for the `ckpt0` to be created, then I launched the eval process again (994 eval episodes).
The model based on depth images returns the correct performances (SPL ~0.95) but unfortunately the one based on RGB images doesn't, and it reports an SPL/SR of about 0.35/0.50 using the `gibson-2plus-mp3d-train-val-test-se-resneXt50-rgb.pth `checkpoint.
Is there something I'm missing? Here there is the config file I'm using:

```
BASE_TASK_CONFIG_PATH: "configs/tasks/pointnav_gibson.yaml"
TRAINER_NAME: "ddppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 0
VIDEO_OPTION: []
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
TEST_EPISODE_COUNT: -1
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
NUM_PROCESSES: 4
SENSORS: ["RGB_SENSOR"]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_UPDATES: 10000
LOG_INTERVAL: 10
CHECKPOINT_INTERVAL: 250

RL:
  SUCCESS_REWARD: 2.5

  POLICY:
    name: "PointNavResNetPolicy"

  PPO:
    # ppo params
    clip_param: 0.2
    ppo_epoch: 2
    num_mini_batch: 2
    value_loss_coef: 0.5
    entropy_coef: 0.01
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.2
    num_steps: 128
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    reward_window_size: 50

    use_normalized_advantage: False

    hidden_size: 512

  DDPPO:
    sync_frac: 0.6
    # The PyTorch distributed backend to use
    distrib_backend: GLOO
    # Visual encoder backbone
    pretrained_weights: data/ddppo-models/gibson-2plus-mp3d-train-val-test-se-resneXt50-rgb.pth #data/ddppo-models/gibson-2plus-resnet50.pth
    # Initialize with pretrained weights
    pretrained: True
    # Initialize just the visual encoder backbone with pretrained weights
    pretrained_encoder: False
    # Whether or not the visual encoder backbone will be trained.
    train_encoder: False
    # Whether or not to reset the critic linear layer
    reset_critic: False

    # Model parameters
    backbone: se_resneXt50 #resnet50
    rnn_type: LSTM
    num_recurrent_layers: 2
```
Here the results for the Depth model:
<pre>
Average episode reward: 7.6915
Average episode distance_to_goal: 0.0944
Average episode success: 0.9960
Average episode spl: 0.9514
</pre>
and the results for the RGB model:
<pre>
Average episode reward: 0.3978                                               
Average episode distance_to_goal: 3.5889                                     
Average episode success: 0.5000                                              
Average episode spl: 0.3489
</pre>

Thank you!

erikwijmans, 2021-01-22:
> For this reason I launched the training process from few seconds, in order for the ckpt0

You can point habitat directly at the checkpoint, just specify it in the `EVAL_CKPT_PATH_DIR:` field.  You then need to add

```
EVAL:
    USE_CKPT_CONFIG: False
```

to the config as those don't have a config baked into them.

rosanom, 2021-01-25:
Thank you @erikwijmans for the quick reply. I took your suggestion, everything worked correctly. It's still unclear what happens when I run the training process starting from the pretrained model, to explain this drop in performances. 

I tried to finetune the ddppo rgb model on the gibson trainset for few frames and then to test the resulting checkpoint on the gibson valset, I get the poor performance reported in the previous message...

Is it possible that the pretrained model is not loaded correctly? What might be the cause of this issue?
Thank you.

StOnEGiggity, 2021-01-25:
Hi, I tested ``gibson-2plus-mp3d-train-val-test-se-resneXt50-rgb.pth`` a few days ago, and the result was similar to that in doc. 

Do you update the habitat-lab and the habitat-sim to the latest version? I remember that the previous version is incompatible with the latest code. 

My habitat-lab commit is ``c19e5666aac3d17ec55f09122ce4acff0873d4b4``, and I install habitat-sim by conda with ``conda install habitat-sim headless -c conda-forge -c aihabitat-nightly``

rosanom, 2021-01-25:
Hi @StOnEGiggity , thank you for your comment. As I said in the previous message, I already installed the latest versions of habitat-lab and habitat-sim. Following @erikwijmans 's last advice I succeded at reproducing the ddppo paper's results if I just launch an evaluation process. Have you ever tried to fine-tune this model on gibson trainset (even just for few frames) and evaluate the navigation performance loading the new generated checkpoint file (in `data/new_checkpoints` folder)? My goal is to train a model in a new environment starting from the one pretrained on gibson/mp3d, but before doing that I just want to make sure that the model is loaded correctly and that it returns the expected performance values when it is used with the original dataset.

erikwijmans, 2021-01-25:
Looks like something was off with the running mean and var state in the checkpoint (the count is zero for some reason).  I have updated that checkpoint to have that fixed.  Fine tuning can still be a little rocky at the start due to the optimizer state, but that gets corrected very fast.

rosanom, 2021-01-27:
Thank you @erikwijmans for your support! 
Just a quick question about the running mean and var task: can you give more details about what it is supposed to do? Does it compute mean and variance of input images all over the trainset, so that they are saved in the checkpoint file and loaded when I decide to resume the training? Thank you!

erikwijmans, 2021-01-27:
Yep

closed 2021-01-28 by rosanom