facebookresearch/habitat-lab #574: No visualization when I run examples/shortest_path_follower_example.py in habitat-lab folder

wwtx9, 2021-01-24:
## ❓ Questions and Help
I have installed habitat-lab and habitat-sim with display, withbullet and no-cuda.
When I run examples/shortest_path_follower_example.py in habitat-lab folder, it shows below without visualization. Did my command is correct?

```
python examples/shortest_path_follower_example.py
2021-01-23 22:51:42,486 Initializing dataset PointNav-v1
2021-01-23 22:51:42,616 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0123 22:51:42.620637  6127 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0123 22:51:42.620672  6127 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0123 22:51:42.620699  6127 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0123 22:51:42.620715  6127 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0123 22:51:42.620723  6127 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0123 22:51:42.620729  6127 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0123 22:51:42.620749  6127 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0123 22:51:42.620766  6127 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0123 22:51:42.620777  6127 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0123 22:51:42.620786  6127 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0123 22:51:42.620798  6127 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0123 22:51:42.620810  6127 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0123 22:51:42.620815  6127 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0123 22:51:42.621174  6127 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) not found so new, default physics manager attributes created and registered.
I0123 22:51:42.621222  6127 StageAttributesManager.cpp:79] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) Based stage attributes created and registered.
I0123 22:51:42.621232  6127 Simulator.cpp:145] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0123 22:51:42.621291  6127 Simulator.cpp:147] Loaded.
I0123 22:51:42.621305  6127 SceneGraph.h:93] Created DrawableGroup: 
Renderer: Mesa DRI Intel(R) UHD Graphics 630 (CML GT2) by Intel Open Source Technology Center
OpenGL version: 4.6 (Core Profile) Mesa 20.0.8
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-implementation-color-read-format-dsa-explicit-binding
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
I0123 22:51:42.631336  6127 ResourceManager.cpp:923] Importing Basis files as ASTC 4x4
I0123 22:51:43.456281  6127 ResourceManager.cpp:302] ResourceManager::loadStage : Not loading semantic mesh
W0123 22:51:43.456303  6127 Simulator.cpp:235] :
---
 The active scene does not contain semantic annotations. 
---
I0123 22:51:43.456539 6127 simulator.py:181] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2021-01-23 22:51:43,457 Initializing task Nav-v0
Environment creation successful
Agent stepping around inside environment.
I0123 22:51:45.466102  6127 SceneGraph.h:93] Created DrawableGroup: 
2021-01-23 22:51:51,777 Video created: examples/images/shortest_path_example/00/trajectory.mp4
  0%|                                                  | 0/77 [00:00<?, ?it/s]IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (599, 256) to (608, 256) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
[swscaler @ 0x66c00c0] Warning: data is not aligned! This can lead to a speed loss
100%|████████████████████████████████████████| 77/77 [00:00<00:00, 796.53it/s]
Episode finished
Agent stepping around inside environment.
2021-01-23 22:51:53,688 Video created: examples/images/shortest_path_example/01/trajectory.mp4
  0%|                                                   | 0/9 [00:00<?, ?it/s]IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (599, 256) to (608, 256) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
[swscaler @ 0x5c3d0c0] Warning: data is not aligned! This can lead to a speed loss
100%|██████████████████████████████████████████| 9/9 [00:00<00:00, 541.40it/s]
Episode finished
Agent stepping around inside environment.
[... 64 lines cut ...]
pyparsing              2.4.7
python-dateutil        2.8.1
PyYAML                 5.4.1
requests               2.25.1
requests-oauthlib      1.3.0
rsa                    4.7
scipy                  1.5.4
setuptools             51.3.3.post20210118
six                    1.15.0
smmap                  3.0.5
tb-nightly             2.5.0a20210123
tensorboard            1.13.1
tensorboard-plugin-wit 1.8.0
tensorflow             1.13.1
tensorflow-estimator   1.13.0
termcolor              1.1.0
torch                  1.7.1
tqdm                   4.56.0
typing-extensions      3.7.4.3
urllib3                1.26.2
Werkzeug               1.0.1
wheel                  0.36.2
yacs                   0.1.8
zipp                   3.4.0
```

erikwijmans, 2021-01-24:
As it says in the logs, it is outputting videos, i.e. `examples/images/shortest_path_example/00/trajectory.mp4`.  If you open that, it should be as expected.

wwtx9, 2021-01-24:
Thanks for your reply!

closed 2021-01-24 by wwtx9