facebookresearch/habitat-lab #576: Extending existing sensors to custom sensors

smorad, 2021-01-26:
## ❓ Questions and Help
I'm trying to add new sensors that leverage current sensors (e.g. a semantic sensor that produces a channel per semantic category). In the past, I would add `[SEMANTIC, MY_SENSOR]` to the configuration file, and use the current semantic observation in my sensor. The problem is the observation then contains the unwanted `SEMANTIC` observations:

```
class MySensor:
...
  def get_observation(self, observations, ...):
    img = observations['semantic']
    # Do stuff with img
    ...

class MySim(habitat.RLEnv):
  def step(...):
    obs = super().step(...)
    obs.pop('semantic')
    ...
    return obs
```

We can remove `semantic` from `obs`, but this requires patching the `observation_space` to any downstream methods. This feels hacky and prone to error.  I was wondering if there was a way to either:

1. Create a sensor that is run at each step, but not registered as an observation and does not affect the `observation_space`
2. Have the step function populate obs with only `TASK` sensors instead of `SIMULATOR` sensors 
3. Render a semantic or RGB image without the use of a `Sensor`

Skylion007, 2021-01-26:
While you can patch custom sensors to do that, we also offer the ObservationTransformer API in Habitat Baselines facilitates this and also allows the observation space to be modified an Env agnostic way. I would recommend using that instead.

smorad, 2021-01-26:
That looks pretty useful. I'll give the following a shot:
1. Write class to inherit from `ObservationTransformer`, implementing logic in `forward`.
2. In `habitat.RLEnv.__init__`, patch observation space using `apply_obs_transforms_obs_space`
3. In `step`, `obs = apply_obs_transforms_batch(obs, transformers)`

smorad, 2021-01-29:
Works like a charm, thanks!

closed 2021-01-29 by smorad