facebookresearch/habitat-lab #584: Continous Action Space?

qingh097, 2021-02-04:
Is it possible to define a customer continuous action space including the linear and angular velocity?

mathfac, 2021-02-07:
@qingh097, thank you for your question.
Related issue https://github.com/facebookresearch/habitat-lab/issues/412. We don't support the linear and angular velocity from Habitat Lab side yet, but it can be added using TaskAction definition with continuous parameters like it's done for Teleport Action [here](https://github.com/facebookresearch/habitat-lab/blob/aa24551fd9942b11b7af4a076d9c05e080594d08/habitat/tasks/nav/nav.py#L1051). In new action you can set velocities in sim like in [this Habitat SIM tutorial](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb#scrollTo=MT9-G1cWgjOa).

rpartsey, 2022-09-04:
Duplicates https://github.com/facebookresearch/habitat-lab/issues/165. 

Continuous actions are very well integrated in Habitat now.

closed 2022-09-04 by rpartsey