facebookresearch/habitat-lab #588: Class NavRLEnv error

aveen-d, 2021-02-05:
## ❓ Questions and Help
I am trying to use class NavRLEnv to train my RL agent. I was successful in loading the config file in the NavRLEnv class. But when I am trying to call env.step function it throws an error. I am attaching the code snippet and error below. 
[image]
[image]

There was no such error up until last week because I was able to train my agent using the same code. Today when I tried to run, I was faced with this error

mathfac, 2021-02-05:
Thank you for the issue. What commit are you using? did you change the code
from original?

aveen-d, 2021-02-05:
No I haven't changed anything from the original code

aveen-d, 2021-02-06:
> Thank you for the issue. What commit are you using? did you change the code from original?
> […](#)

Hi, is there any update?

erikwijmans, 2021-02-06:
That line of code hasn't changed in 17 months: https://github.com/facebookresearch/habitat-lab/blame/master/habitat_baselines/common/environments.py#L53 so it's hard to figure out what changed.  Unless you were on a version from more than a year and a half ago, there isn't any way were the same code would have worked a week ago.

Perhaps you were using habitat.Env instead of going through NavRLEnv?

aveen-d, 2021-02-07:
> That line of code hasn't changed in 17 months: https://github.com/facebookresearch/habitat-lab/blame/master/habitat_baselines/common/environments.py#L53 so it's hard to figure out what the issue is. Unless you were on a version from more than a year and a half ago, there isn't any way were the same code would have worked a week ago.
> 
> Perhaps you were using habitat.Env instead of going through NavRLEnv?

Can you please help me with this. I want to use the NavRLEnv ( the defined reward function, step and reset function) to train my custom RL agent. I loaded the config file as shown above in the NavRLEnv but env.step(action) throws the same error as above. How can I resolve this?

aveen-d, 2021-02-07:
The env is loaded correctly because env.reset() function works and gives me the output of state observation, only env.step isn't working

aveen-d, 2021-02-07:
from the code what I observe is only the step function is using '*args and **kwargs', do you think is that causing some issue? Because other functions which doesn't use it works properly.
[image]

erikwijmans, 2021-02-07:
`env.step(action=dict(action="STOP"))`

dhruvbatra, 2021-02-08:
In general, please try to stick to a release version. 
https://github.com/facebookresearch/habitat-sim/releases

aveen-d, 2021-02-08:
> `env.step(action=dict(action="STOP"))`

It worked with this. Thank you sir

aveen-d, 2021-02-08:
> In general, please try to stick to a release version.
> https://github.com/facebookresearch/habitat-lab/releases

Okay sir thanks for the help

closed 2021-02-08 by aveen-d