facebookresearch/habitat-lab #597: VectorEnv/ThreadedVectorEnv have worse performance than Env

frasermcghan, 2021-02-21:
## ❓ Questions and Help

Hi,

I wrote the following script to compare the performance of different types of habitat Envs.

I expected the vectorised environments to run faster than the normal Env, however  got the following results:

```terminal
Normal Env took 299.7s to take 500 steps in 14 environments
VectorEnv took 484.4s to take 500 steps in 14 environments
ThreadedVectorEnv took 2345.4s to take 500 steps in 14 environments
```

Can anyone explain this result please? 

BTW: This was run on an RTX 2080Ti GPU.

```python
"""
example.py
==================================================
This is an example of how to use the habitat environment class.
"""

import habitat
from habitat_baselines.utils.env_utils import (
    make_env_fn,
    construct_envs,
    construct_threaded_envs,
)
from habitat_baselines.config.default import get_config

def example():
    """
    Initialises a `habitat.Env` environment, then takes random steps.
    """

    baseline_config = get_config("configs/tasks/pointnav.yaml")
    config = habitat.get_config("configs/tasks/pointnav.yaml")

    config.defrost()
    config.TASK_CONFIG = baseline_config.TASK_CONFIG
    config.freeze()

    env_class = habitat.Env

    env = make_env_fn(config, env_class)

    n_steps = 500
    n_episodes = env.number_of_episodes

    for episode in range(n_steps):
        count_steps = 0
        _ = env.reset()

        scene_name = env.current_episode.scene_id.split(sep="/")[-1].strip(".glb")
        print(f"\nStarting episode {episode+1}/{n_episodes} in {scene_name}...")

        print("Stepping around...")
        for step in range(n_steps):
            action = env.action_space.sample()

            while action["action"] == "STOP":
                action = env.action_space.sample()

            _ = env.step(action)
            count_steps += 1

        print(
            f"Episode {episode+1} in scene {scene_name} finished after {count_steps} steps."
        )

def vector_example():
    """
    Initialises a `habitat.VectorEnv` environment, then takes random steps.
    """

    baseline_config = get_config("configs/tasks/pointnav.yaml")
    config = habitat.get_config("configs/tasks/pointnav.yaml")

    config.defrost()
    config.TASK_CONFIG = baseline_config.TASK_CONFIG
    config.NUM_ENVIRONMENTS = 14
    config.SENSORS = baseline_config.SENSORS
    config.SIMULATOR_GPU_ID = 0
    config.freeze()

    env_class = habitat.Env
    envs = construct_envs(config, env_class)

    _ = envs.reset()

    n_steps = 500
    n_envs = envs.num_envs
    n_episodes = envs.number_of_episodes[0]

    for episode in range(n_episodes):
        count_steps = 0
        print(
            f"Starting episode {episode}/{n_episodes} in parallel processes across {n_envs} environments."
        )
        scene_names = [
            ep.scene_id.split("/")[-1].strip(".glb") for ep in envs.current_episodes()
        ]

        for i, scene in enumerate(scene_names):
            print(f"Stepping around in Environment {i+1}: {scene}..")

        for step in range(n_steps):
            random_actions = [
                action_space.sample() for action_space in envs.action_spaces
            ]
            _ = envs.step(random_actions)
            count_steps += 1

        print("Episodes finished after {} steps.".format(count_steps))

def threaded_vector_example():
    """
    Initialises a `habitat.ThreadedVectorEnv` environment, then takes random steps.
    """

    baseline_config = get_config("configs/tasks/pointnav.yaml")
    config = habitat.get_config("configs/tasks/pointnav.yaml")

    config.defrost()
    config.TASK_CONFIG = baseline_config.TASK_CONFIG
    config.NUM_ENVIRONMENTS = 14
    config.SENSORS = baseline_config.SENSORS
    config.SIMULATOR_GPU_ID = 0
    config.freeze()

    env_class = habitat.Env

    envs = construct_threaded_envs(config, env_class)

    _ = envs.reset()

    n_steps = 500
    n_envs = envs.num_envs
    n_episodes = envs.number_of_episodes[0]

    for episode in range(n_episodes):
        count_steps = 0
        print(
            f"Starting episode {episode}/{n_episodes} in parallel threads across {n_envs} environments."
        )
        scene_names = [
            ep.scene_id.split("/")[-1].strip(".glb") for ep in envs.current_episodes()
        ]

        for i, scene in enumerate(scene_names):
            print(f"Stepping around in Environment {i+1}: {scene}..")

        for step in range(n_steps):
            random_actions = [
                action_space.sample() for action_space in envs.action_spaces
            ]
            _ = envs.step(random_actions)
            count_steps += 1

        print("Episodes finished after {} steps.".format(count_steps))

if __name__ == "__main__":
    import time

    start = time.time()
    example()
    end = time.time()
    example_time = end - start

    start = time.time()
    vector_example()
    end = time.time()
    vector_example_time = end - start

    start = time.time()
    threaded_vector_example()
    end = time.time()
    threaded_example_time = end - start

    print(
        f"\n\n\nNormal Env took {round(example_time, 1)}s to take 500 steps in 14 Environments"
    )
    print(
        f"VectorEnv took {round(vector_example_time, 1)}s to take 500 steps in 14 Environments"
    )
    print(
        f"ThreadedVectorEnv took {round(threaded_example_time, 1)}s to take 500 steps in 14 Environments"
    )
```

dhruvbatra, 2021-02-22:
CC: @Skylion007

Skylion007, 2021-02-22:
The first loop should be on n_episodes like the others. This script is not a fair comparison.

erikwijmans, 2021-02-22:
Also, I'll clarify that `ThreadedVectorEnv` is not actually parallel (it uses python threads).  That exists solely for debugging purposes and it being slow is a (necessary, albeit unfortunate) side effect of it fulfilling that role.  It is not intended for use when performance matters.

frasermcghan, 2021-02-22:
Thanks both for the comments 👍 

@Skylion007 I have changed the n_steps typo and Env now takes 605.7s so that makes sense! 

Thanks for the info @erikwijmans, although I think there may be some ambiguity in the docstring for ThreadedVectorEnv, shouldn't it say "This makes it much easier to debug when using *`ThreadedVectorEnv`* because you can actually put break points in the environment methods"? 

```python
class ThreadedVectorEnv(VectorEnv):
    r"""Provides same functionality as :ref:`VectorEnv`, the only difference
    is it runs in a multi-thread setup inside a single process.
    The :ref:`VectorEnv` runs in a multi-proc setup. This makes it much easier
    to debug when using :ref:`VectorEnv` because you can actually put break
    points in the environment methods. It should not be used for best
    performance.
    """
```

erikwijmans, 2021-02-22:
I think `than when using` is what was intended there.

The timing here still doesn't make sense.  That would imply 14 processes only gets you 1.25x faster.  

How are you doing episode splitting (I ask because the default splitting is does it per scene, but pointnav.yaml doesn't have 14 scenes, so I assume you are doing something else)?  In general, VectorEnv is not optimized for going through a fixed number of episodes as fast as possible, it's optimized for generating experience assuming you want to loop over the set of episodes.  When a process finishes it's episodes it simply restarts its loop over them, so unless you've split episodes evenly between processes VectorEnv is still doing more work.

One other thought:  Perhaps this timing does make sense.  If you aren't loading the simple example scenes but are loading Gibson or MP3D, 14 scenes will easily exceed the VRAM on a 2080Ti.  OpenGL will then resort to streaming textures from CPU memory as needed, which is much slower.

rpartsey, 2022-09-05:
Closing the issue because it has not had recent activity.

closed 2022-09-05 by rpartsey