facebookresearch/habitat-lab #601: VectorEnv Broken Pipe

NaveenVSrinivasan, 2021-02-24:
## 🐛 Bug

For the purpose of debugging/better support, I'm switching from using ThreadedVectorEnv to using VectorEnv for my training multiprocessing. When I use the **exact same** config/env_class that I was passing into construct_envs_threaded (that functioned with no errors) into construct_envs, I get a broken pipe error which seems to indicate that one of my subprocesses is segfaulting. I was told that this could be caused because OpenGL/CUDA are not fork safe, but when I try to directly set the forking context (using `mp.set_start_method('forkserver', force=True)`), I get the following error.

RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.
        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:
            if __name__ == '__main__':
                freeze_support()
                ...
        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.

## Additional context

I am executing my code on an EC2 instance running Ubuntu 18.04 (Bionic Beaver).

erikwijmans, 2021-02-25:
If you were successfully using `ThreadedVectorEnv` then fork vs. forkserver isn't the issue.  Running `dmesg` after the crash can be helpful as that'll show the reason why the processes died.  That often isn't specific enough to fix the issue, but it'll tell us a place to start.

NaveenVSrinivasan, 2021-02-25:
I'm getting several statements that say:
`[11023390.374576] nvidia-modeset: WARNING: GPU:0: Correcting number of heads for current head configuration (0x00)`

The full output of dmesg is attached here.
[error.txt](https://github.com/facebookresearch/habitat-lab/files/6044716/error.txt)

erikwijmans, 2021-02-25:
I've never seen that issue before.  Is this a machine with an attached display?  Either way, my guess would be that that is a hardware issue.

NaveenVSrinivasan, 2021-02-25:
This is running on an EC2 instance, so no attached display. The issue persists when I build habitat-sim from source using setup.py with the --headless flag.

erikwijmans, 2021-02-25:
This isn't something I have seen before, so I don't know what to do with it.  One other thing that may be helpful is to turn on verbose logging from Magnum:  
```
export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON GLOG_minloglevel=0
```

NaveenVSrinivasan, 2021-02-25:
I turned on verbose logging from Magnum and dmesg is now giving us more information on the seg fault. 
[error.txt](https://github.com/facebookresearch/habitat-lab/files/6046197/error.txt)

erikwijmans, 2021-02-25:
If you can give me a at least some way to reproduce I can try to figure it out, but without one there isn't much I can do unless there is something in the logging -- not from dmesg, just from what the simulator prints

rpartsey, 2022-09-05:
Hi @NaveenVSrinivasan have you managed to resolve the issue described above?

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity.