facebookresearch/habitat-lab #611: Unique seed not set for each simulator

Tushar-N, 2021-03-17:
Habitat-Lab: v0.1.7 & master

## 🐛 Bug

Each environment should get a unique seed, but it looks like each process in the process group is provided the same seed in this code: https://github.com/facebookresearch/habitat-lab/blob/a7eb39fb3b420f9e6f528796f97457715bd79a09/habitat_baselines/rl/ppo/ppo_trainer.py#L217
```
# Multiply by the number of simulators to make sure they also get unique seeds
self.config.TASK_CONFIG.SEED += (
    torch.distributed.get_world_size()
    * self.config.NUM_ENVIRONMENTS
)
```

## Expected behavior
As far as I understand, `torch.distributed.get_world_size()` is a fixed number (the number of processes in the process group). Instead, shouldn't `local_rank` be used to modify the seed?
```
# Multiply by the number of simulators to make sure they also get unique seeds
self.config.TASK_CONFIG.SEED += (
    local_rank
    * self.config.NUM_ENVIRONMENTS
)
```

dhruvbatra, 2021-03-17:
CC: @mathfac

erikwijmans, 2021-03-17:
Yeah, that should be `torch.distributed.get_rank()`.  Made a PR

PR #612 (2021-03-25): Fix little bug in seeding
## Motivation and Context

Closes #611
closed 2021-03-25 by erikwijmans