facebookresearch/habitat-lab #613: register `ObjectNav` environment without installing full `habitat_baselines`

ethanabrooks, 2021-03-17:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7

## ❓ Questions and Help
Hi. I would like to use the environment corresponding to [ObjectNav](https://aihabitat.org/challenge/2021/#task-2-objectnav). 
My first question is, does that correspond to [`NavRLEnv`](class NavRLEnv(habitat.RLEnv))?
Second is there a way to import just this environment (or ideally the registered version of it) without installing `tensorflow` and all the other machine learning dependencies? If not I am happy to work on a pull request.

Skylion007, 2021-03-18:
Yeah, a pull request to have a better setup.py would be greatly appreciated and we would definitely welcome that PR.

Our current installation standard is very off-spec, so a more pythonic install would definitely be greatly appreciated. It would be great if the setup.py conformed the pip extras_features so that you could ```pip install habitat-lab[full]``` or ```pip install habitat-lab[baselines]``` like so: https://stackoverflow.com/questions/6237946/optional-dependencies-in-distutils-pip 

As a temporary work around, you can do ```pip install -e .``` on the git root. That should install it without installing the requirements.

ethanabrooks, 2021-03-18:
I noticed that the Dockerfile does not currently build:
```
executor failed running [/bin/sh -c curl -o ~/miniconda.sh -O  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh  &&    chmod +x ~/miniconda.sh &&    ~/miniconda.sh -b -p /opt/conda &&    rm ~/miniconda.sh &&    /opt/conda/bin/conda install numpy pyyaml scipy ipython mkl mkl-include &&    /opt/conda/bin/conda clean -ya]: exit code: 127
```
Would you like me to include a fix in this pull request?

ethanabrooks, 2021-03-18:
Oh no I was just using the Dockerfile in the root of the project.

mathfac, 2021-03-18:
To clarify, for training you’re not necessary need Docker.
Docker is need to send your agent for submission.
If you’re preparing agent for submission it’s expected to use Dockers from
https://github.com/facebookresearch/habitat-challenge/ repo.

ethanabrooks, 2021-03-18:
I should probably mention, my approach is going to be to use `python setup.py test` to ensure that my changes don't break anything. I'm also going to try to make sure that the installation methods that I write work in docker.

ethanabrooks, 2021-03-18:
> To clarify, for training you’re not necessary need Docker.
Docker is need to send your agent for submission.
If you’re preparing agent for submission it’s expected to use Dockers from
https://github.com/facebookresearch/habitat-challenge/ repo.

I'm not planning to submit an agent for submission. I'm just planning to use Docker to make sure the build works in a consistent development environment.

ethanabrooks, 2021-03-21:
So I ended up not needing to create the pull request, but I thought I should post my solution for anyone else wishing to train on the ObjectNav environment but not interested in submitting to the Habitat Challenge:
```python
class NavRLEnv(habitat.RLEnv):
    def __init__(self, config, rank: int):
        config.defrost()
        config.SEED += rank
        gpus = GPUtil.getAvailable(limit=math.inf)
        if gpus:
            # put each environment on a different GPU
            config.SIMULATOR.HABITAT_SIM_V0.GPU_DEVICE_ID = int(gpus[rank % len(gpus)])
        self._config = config
        config.freeze()

        # Config the dataset with given scene and object_category
        dataset = habitat.make_dataset(
            id_dataset=self._config.DATASET.TYPE, config=self._config.DATASET
        )
        super().__init__(config=self._config, dataset=dataset)

    def get_reward_range(self):
        return (
            self._config.RL.SLACK_REWARD,
            self._config.RL.SUCCESS_REWARD,
        )

    def get_reward(self, observations):
        return (
            self._config.RL.SUCCESS_REWARD
            if self._episode_success()
            else self._config.RL.SLACK_REWARD
        )

    def _episode_success(self):
        return self._env.get_metrics()[self._config.RL.SUCCESS_MEASURE]

    def get_done(self, observations):
        return self._env.episode_over or self._episode_success()

    def get_info(self, observations):
        return self.habitat_env.get_metrics()
```
Note that we inherit from `RLEnv` not `Env`. `RLEnv` is a wrapper around `Env` that conforms to the `gym.Env` interface.
For the config, I use [`challenge_objectnav2021.local.rgbd.yaml`:](https://github.com/facebookresearch/habitat-challenge/blob/master/configs/challenge_objectnav2021.local.rgbd.yaml)
```yaml
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
  ITERATOR_OPTIONS:
    SHUFFLE: False
SIMULATOR:
  TURN_ANGLE: 30
  TILT_ANGLE: 30
  ACTION_SPACE_CONFIG: "v1"
  AGENT_0:
    SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR']
    HEIGHT: 0.88
    RADIUS: 0.18
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
    ALLOW_SLIDING: False
  SEMANTIC_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    POSITION: [0, 0.88, 0]
  RGB_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    POSITION: [0, 0.88, 0]
  DEPTH_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    MIN_DEPTH: 0.5
    MAX_DEPTH: 5.0
    POSITION: [0, 0.88, 0]
TASK:
  TYPE: ObjectNav-v1
  POSSIBLE_ACTIONS: ["STOP", "MOVE_FORWARD", "TURN_LEFT", "TURN_RIGHT", "LOOK_UP", "LOOK_DOWN"]
  SUCCESS_DISTANCE: 0.1

  SENSORS: ['OBJECTGOAL_SENSOR', 'COMPASS_SENSOR', 'GPS_SENSOR']
  GOAL_SENSOR_UUID: objectgoal

  MEASUREMENTS: ['DISTANCE_TO_GOAL', 'SUCCESS', 'SPL', 'SOFT_SPL']

  DISTANCE_TO_GOAL:
    DISTANCE_TO: VIEW_POINTS
  SUCCESS:
    SUCCESS_DISTANCE: 0.1

DATASET:
  TYPE: ObjectNav-v1
  SPLIT: val_mini
  DATA_PATH: habitat-challenge-data/objectgoal_mp3d/{split}/{split}.json.gz
  SCENES_DIR: "habitat-challenge-data/data/scene_datasets/"
 ```
For downloading the data, I followed [these instruction](https://github.com/facebookresearch/habitat-challenge#pointnavobjectnav-baselines-and-dd-ppo-training-starter-code).

closed 2021-03-21 by ethanabrooks