facebookresearch/habitat-lab #616: Questions about agent's heading

JeremyLinky, 2021-03-29:
I've computed the angle like that before by adding `pi/2` to the negative angle but it doesn't seem to be right.
For example, when the agent turns left by `pi/6`, the pose `(x, y, theta)` changes from `(7.35, -4.61, 0.74)` to `(7.35, -4.61, 0.21)`, which should be `(7.35, -4.61, 0.74 + pi/6) = (7.35, -4.61, 1.25)` if the computation is right.
When the agent moves forward by `0.3`, the agent moves from `(6.76, -4.74, 0.21)` to `(6.47, -4.80, 0.21)`, which is not right as the correct next pose should be `(6.76 + 0.3*cos(0.21), -4.74 + 0.3*sin(0.21), 0.21) = (7.06, -4.73, 0.21)` if the computation is right.

Sorry to bother you so much.
And I really don't understand the logic of the conversion to angle. Why do we add pi/2 to the final result?

_Originally posted by @yimengli46 in https://github.com/facebookresearch/habitat-lab/issues/247#issuecomment-551224326_

erikwijmans, 2021-03-31:
The heading sensor returns 0 when the agent is looking down -Z and pi/2 when its looking down Y.

In that issue, I pointed out that the pi/2 thing is solving a different issue.  I think for that one the resolution was to add pi as that is accounting for the 180 degree flip that happens since they weren't negating Z.

rpartsey, 2022-09-05:
Closing the issue because it has not had recent activity.

closed 2022-09-05 by rpartsey