facebookresearch/habitat-lab #617: TypeError: write_gif() got an unexpected keyword argument 'verbose'

jsdd25, 2021-03-29:
## Habitat-Lab version
Habitat-Lab: master

## ❓ Questions and Help
Greetings,
When I run `python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval`, I met the following problems:
`(habitat7) jin@jin-MS-7B94:~/habitat/habitat-lab$ python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval
I0329 14:56:31.935843 7802 simulator.py:213] Loaded navmesh data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
2021-03-29 14:56:31,937 Initializing task Nav-v0
  1%|▍                                       | 1/100 [04:29<7:23:55, 269.05s/it]2021-03-29 15:01:05,585 Video created: video_dir/episode=7-ckpt=0-distance_to_goal=1.75-success=0.00-spl=0.00-collisions.count=5.00.mp4
                                                                               IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (1002, 256) to (1008, 256) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
                                                                               [swscaler @ 0x66daac0] Warning: data is not aligned! This can lead to a speed loss
100%|████████████████████████████████████████| 499/499 [00:00<00:00, 511.14it/s]
Traceback (most recent call last):████████▍  | 467/499 [00:00<00:00, 122.77it/s]
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/site-packages/torch/utils/tensorboard/summary.py", line 467, in make_video
    clip.write_gif(filename, verbose=False, logger=None)
TypeError: write_gif() got an unexpected keyword argument 'verbose'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/site-packages/torch/utils/tensorboard/summary.py", line 470, in make_video
    clip.write_gif(filename, verbose=False, progress_bar=False)
TypeError: write_gif() got an unexpected keyword argument 'verbose'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/home/jin/habitat/habitat-lab/habitat_baselines/common/base_trainer.py", line 128, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/home/jin/habitat/habitat-lab/habitat_baselines/rl/ppo/ppo_trainer.py", line 651, in _eval_checkpoint
    tb_writer=writer,
  File "/home/jin/habitat/habitat-lab/habitat_baselines/utils/common.py", line 215, in generate_video
    f"episode{episode_id}", checkpoint_idx, images, fps=fps
  File "/home/jin/habitat/habitat-lab/habitat_baselines/common/tensorboard_utils.py", line 67, in add_video_from_np_images
    video_name, video_tensor, fps=fps, global_step=step_idx
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/site-packages/torch/utils/tensorboard/writer.py", line 666, in add_video
    video(tag, vid_tensor, fps), global_step, walltime)
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/site-packages/torch/utils/tensorboard/summary.py", line 442, in video
    video = make_video(tensor, fps)
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/site-packages/torch/utils/tensorboard/summary.py", line 472, in make_video
    clip.write_gif(filename, verbose=False)
TypeError: write_gif() got an unexpected keyword argument 'verbose'
  1%|▍                                       | 1/100 [04:32<7:29:01, 272.14s/it]
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7fcdf76e2a20>>
Traceback (most recent call last):
  File "/home/jin/habitat/habitat-lab/habitat/core/vector_env.py", line 563, in __del__
  File "/home/jin/habitat/habitat-lab/habitat/core/vector_env.py", line 429, in close
  File "/home/jin/anaconda3/envs/habitat7/lib/python3.6/multiprocessing/connection.py", line 206, in send
AttributeError: 'NoneType' object has no attribute 'dumps'
`
It seems to be an issue in the tensorboard or tensorflow. My GPU is GTX 3090. The version of my tensorflow is 2.0.0 and the version of my tensorboard is 2.4.1. 
Thanks

jsdd25, 2021-03-29:
I have fixed it by installing the moviepy package with the commond: `conda install -c conda-forge moviepy`

closed 2021-03-29 by jsdd25

xwaiyy123, 2022-06-28:
> I have fixed it by installing the moviepy package with the commond: `conda install -c conda-forge moviepy`
I executed your command but I got a new error：
data/scene_datasets/habitat-test-scenes/van-gogh-room.glb
[10:22:57:135039]:[Sim] Simulator.cpp(471)::instanceStageForSceneAttributes : 
---
The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0  
---
[10:22:57:135084]:[Sim] Simulator.cpp(205)::reconfigure : CreateSceneInstance success == true for active scene name : data/scene_datasets/habitat-test-scenes/van-gogh-room.glb  with renderer.
2022-06-28 10:22:57,147 Initializing task Nav-v0
 50%|██████████████████████████████████████▌                                      | 1/2 [00:04<00:04,  4.93s/it]2022-06-28 10:23:10,893 Video created: video_dir/episode=7-ckpt=0-distance_to_goal=2.30-success=0.00-spl=0.00-collisions.count=0.00.mp4
                                                                                                               IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (1002, 256) to (1008, 256) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
100%|█████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  2.80it/s]
[swscaler @ 0x5630178d6dc0] Warning: data is not aligned! This can lead to a speed loss [00:00<00:00,  2.80it/s]
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 81, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 77, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 62, in execute_exp
    trainer.eval()
  File "/test/datasets/habitat-lab-main/habitat_baselines/common/base_trainer.py", line 130, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/test/datasets/habitat-lab-main/habitat_baselines/rl/ppo/ppo_trainer.py", line 1108, in _eval_checkpoint
    keys_to_include_in_name=self.config.EVAL_KEYS_TO_INCLUDE_IN_NAME,
  File "/test/datasets/habitat-lab-main/habitat_baselines/utils/common.py", line 402, in generate_video
    f"episode{episode_id}", checkpoint_idx, images, fps=fps
  File "/test/datasets/habitat-lab-main/habitat_baselines/common/tensorboard_utils.py", line 86, in add_video_from_np_images
    video_name, video_tensor, fps=fps, global_step=step_idx
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/torch/utils/tensorboard/writer.py", line 666, in add_video
    video(tag, vid_tensor, fps), global_step, walltime)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/torch/utils/tensorboard/summary.py", line 442, in video
    video = make_video(tensor, fps)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/torch/utils/tensorboard/summary.py", line 467, in make_video
    clip.write_gif(filename, verbose=False, logger=None)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/decorators.py", line 54, in requires_duration
    return f(clip, *a, **k)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/decorators.py", line 22, in convert_masks_to_RGB
    return f(clip, *a, **k)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/video/VideoClip.py", line 434, in write_gif
    logger=logger)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/video/io/gif_writers.py", line 283, in write_gif_with_image_io
    for frame in clip.iter_frames(fps=fps, logger=logger, dtype='uint8'):
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/decorators.py", line 54, in requires_duration
    return f(clip, *a, **k)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/root/anaconda3/envs/habitat/lib/python3.7/site-packages/moviepy/decorators.py", line 118, in use_clip_fps_by_default
    names = inspect.getfullargspec(func).args[1:]
NameError: name 'func' is not defined
 50%|██████████████████████████████████████▌                                      | 1/2 [00:06<00:06,  6.18s/it]
Exception ignored in: <function VectorEnv.__del__ at 0x7f74d3c455f0>
Traceback (most recent call last):
  File "/test/datasets/habitat-lab-main/habitat/core/vector_env.py", line 592, in __del__
    self.close()
  File "/test/datasets/habitat-lab-main/habitat/core/vector_env.py", line 463, in close
    write_fn((CLOSE_COMMAND, None))
  File "/test/datasets/habitat-lab-main/habitat/core/vector_env.py", line 118, in __call__
    self.write_fn(data)
  File "/test/datasets/habitat-lab-main/habitat/utils/pickle5_multiprocessing.py", line 62, in send
    self.send_bytes(buf.getvalue())
  File "/root/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 200, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/root/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/root/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

luckeciano, 2022-08-03:
The root cause for this problem is that the `requirements.txt` in `habitat_baselines/rl` requires `moviepy>=1.0.1`, and moviepy had a breaking change (the current version is `2.x`). An easy way to overcome such problems is to fix the version to install:

`pip install moviepy==1.0.1`

As a long-term solution, we should at least add this constraint to the `requirements.txt` file.