facebookresearch/habitat-lab #621: Error occurs when running "examples/vln_reference_path_follower_example.py"

EricWZH, 2021-04-06:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.1.7

Habitat-Sim: 0.1.7

## ❓ Questions and Help
I have installed habitat-lab and habitat-sim on headless machines. I can run examples/shortest_path_follower_example.py successfully. However,when I run examples/vln_reference_path_follower_example.py in habitat-lab folder, it breaks and shows below. I wonder how to solve the issue?
```
2021-04-06 17:46:13,094 Initializing dataset R2RVLN-v1
2021-04-06 17:46:13,166 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0406 17:46:13.167728  5814 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
I0406 17:46:13.184638  5814 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0406 17:46:13.184926  5814 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0406 17:46:13.185039  5814 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0406 17:46:13.185156  5814 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0406 17:46:13.185235  5814 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0406 17:46:13.185281  5814 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0406 17:46:13.185315  5814 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0406 17:46:13.185436  5814 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0406 17:46:13.185531  5814 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0406 17:46:13.185583  5814 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0406 17:46:13.185627  5814 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0406 17:46:13.185693  5814 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0406 17:46:13.185752  5814 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0406 17:46:13.185779  5814 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0406 17:46:13.187259  5814 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I0406 17:46:13.187273  5814 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
I0406 17:46:13.234864  5814 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  does not exist.
I0406 17:46:13.234930  5814 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
I0406 17:46:13.234944  5814 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb
I0406 17:46:13.234954  5814 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb changed to proposed JSON configuration filename : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.stage_config.json
I0406 17:46:13.234966  5814 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.stage_config.json from original name : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb | This file  does not exist.
I0406 17:46:13.235090  5814 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb
I0406 17:46:13.235097  5814 AbstractObjectAttributesManagerBase.h:188] File (data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I0406 17:46:13.235116  5814 Simulator.cpp:156] Loading navmesh from data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.navmesh
I0406 17:46:13.271909  5814 Simulator.cpp:158] Loaded.
I0406 17:46:13.279351  5814 SceneGraph.h:93] Created DrawableGroup: 
Renderer: PH402 SKU 200/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64.00
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0406 17:46:13.528337  5814 ResourceManager.cpp:234] ResourceManager::loadStage : Not loading semantic mesh
I0406 17:46:13.528364  5814 ResourceManager.cpp:262] ResourceManager::loadStage : start load render asset data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb.
I0406 17:46:13.528370  5814 ResourceManager.cpp:569] ResourceManager::loadStageInternal : Attempting to load stage data/scene_datasets/mp3d/aayBHfsNo7d/aayBHfsNo7d.glb 
I0406 17:46:13.528396  5814 ResourceManager.cpp:1119] Importing Basis files as BC7 for aayBHfsNo7d.glb
W0406 17:46:25.228698  5814 Simulator.cpp:248] :
---
 The active scene does not contain semantic annotations. 
---
[... 67 lines cut ...]
protobuf             4.0.0rc2
pycparser            2.20
pyglet               1.5.0
pyparsing            2.4.7
python-dateutil      2.8.1
PyYAML               5.4.1
requests             2.25.1
scipy                1.5.4
setuptools           52.0.0.post20210125
simplejson           3.17.2
six                  1.15.0
smmap                3.0.5
tb-nightly           2.5.0a20210401
tensorboard          1.13.1
tensorflow           1.13.1
tensorflow-estimator 1.13.0
termcolor            1.1.0
torch                1.3.1
tornado              6.1
tqdm                 4.59.0
webdataset           0.1.40
wheel                0.36.2
yacs                 0.1.8
yapf                 0.31.0
```

Wangnaijia, 2021-10-13:
I've met the same question, have you solve this?

rpartsey, 2023-01-26:
Fixed in scope of https://github.com/facebookresearch/habitat-lab/pull/983.

closed 2023-01-26 by rpartsey