facebookresearch/habitat-lab #623: Confusion about training speed between different habitat-lab version

StOnEGiggity, 2021-04-10:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.3 and v0.1.6
Habitat-Sim: v0.1.7

Hi, I am updating my codebase based on habitat-lab v0.1.3 (commit 6b80fc4ca5edf14a161a9f6482d875ec12a913a0)to match the latest version (v0.1.6 commit aa24551fd9942b11b7af4a076d9c05e080594d08). However, I find the training speed decreased from about 200 FPS to 25 FPS. The code is modified from the previous baseline code without DDP. And I only change some interfaces to match the latest version.

I know the above description may not be enough, and it's hard to track the problem. Hopefully, I want to know the possible reason. Thank you very much.

erikwijmans, 2021-04-12:
If you also updated pytorch, you may have hit some changes in defaults there that reduced perf here, try updating to https://github.com/facebookresearch/habitat-lab/pull/595 or pulling in #595 and #594.

StOnEGiggity, 2021-04-13:
Thanks for your reply.

closed 2021-04-13 by StOnEGiggity