facebookresearch/habitat-lab #631: Segmentation fault core dumped when running python example script

tanhl2000, 2021-04-21:
Hi, trying out AI Habitat, got this error after running the python example scipt:

-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
sim_cfg.physics_config_file = ./data/default.physics_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0421 20:32:50.211889 22390 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
I0421 20:32:50.211931 22390 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0421 20:32:50.212020 22390 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0421 20:32:50.212081 22390 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0421 20:32:50.212141 22390 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0421 20:32:50.212184 22390 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0421 20:32:50.212203 22390 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0421 20:32:50.212220 22390 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0421 20:32:50.212276 22390 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0421 20:32:50.212319 22390 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0421 20:32:50.212345 22390 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0421 20:32:50.212369 22390 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0421 20:32:50.212401 22390 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0421 20:32:50.212430 22390 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0421 20:32:50.212445 22390 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0421 20:32:50.213223 22390 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I0421 20:32:50.213234 22390 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
I0421 20:32:50.218580 22390 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I0421 20:32:50.218652 22390 PhysicsAttributesManager.cpp:27] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I0421 20:32:50.218664 22390 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
I0421 20:32:50.218677 22390 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb changed to proposed JSON configuration filename : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.stage_config.json
I0421 20:32:50.218688 22390 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.stage_config.json from original name : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb | This file  does not exist.
E0421 20:32:50.218778 22390 StageAttributesManager.cpp:90] StageAttributesManager::registerObjectFinalize : Render asset template handle : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb specified in stage template with handle : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb does not correspond to any existing file or primitive render asset.  Aborting. 
I0421 20:32:50.218794 22390 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
Segmentation fault (core dumped)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------

System details: Ubuntu 20.0.4, GTX 920M and nvidia driver 460.

Wondering if anyone has any solutions, thank you!

aclegg3, 2021-04-21:
> /data/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb | This file does not exist.

Looks like you need to download the example assets as described in the [Testing](https://github.com/facebookresearch/habitat-sim#testing) section of the Habitat-sim README.

tanhl2000, 2021-04-22:
Hi aclegg3,

Thanks for your help, it was the leading / causing the issue.

closed 2021-04-22 by tanhl2000