facebookresearch/habitat-lab #632: Question about rotations in Embodied Question Answering task dataset

YasMME, 2021-04-26:
I wanted to check if the rotations in the embodied question answering matterport task dataset are in the format: **[w, x, y, x]**? Looking at the values, that seems to make sense, but I saw in another discussion of rotations that habitat uses [x, y, z, w], so I'm unsure.

erikwijmans, 2021-04-26:
No, they are also in `[x, y, z, w]`

YasMME, 2021-04-27:
Thank you for the quick reply! I'm still a little confused about interpreting them--haven't worked with quaternions before. Would they correspond to `a + bi + cj + dk` as `w = a, x = b, y = c, z = d`? And i,j,k are unit vectors from a global origin defined by the mesh? or would it be `cos(w/2) + sin(w/2)(xi+yj+zk)`?

erikwijmans, 2021-04-27:
The former.  However, the `i,j,k` unit vectors are in habitat-sim's coordinate frame, which is +Y up and -Z forward while the MP3D meshes are +Z up (they don't have a defined notion of forward but +Y gets mapped to -Z)