facebookresearch/habitat-lab #633: Question about VQA/PACMAN config files

nilinykh, 2021-04-28:
Hi,

I have been looking at the config files for the EQA files and noticed the following differences:

in `il_pacman_nav.yaml` it says that
> SENSORS: ["RGB_SENSOR", "DEPTH_SENSOR"]

while in `il_vqa.yaml` it says that
> SENSORS: ["RGB_SENSOR"]

I was not able to track whether 'depth_sensor' is somehow used to get data for pacman, so does this difference in config files influence training/evaluation of the navigator and the vqa module?
From the pacman and vqa code (and from the EQA paper), I can see that the visual encoder is taking image_feat, and it's unclear to me whether these are only RGB values or some type of RGB + Depth vector for pacman. For VQA, these are only RGB values, I assume.

Does this difference in SENSORS affect the input to pacman and vqa model in any way?

Best,
Nikolai.

mukulkhanna, 2021-06-04:
Hey @nilinykh, sorry for the late response. Please feel free to tag me for any future queries related to the EQA implementation. 

Sorry for the condusion. The depth sensor made available using the config file seems to be a small mistake. However, it does not affect the input, training, or evaluation in any way -- only the RGB images are fed to the network.

Having the depth sensor in the config only makes the depth observations available. However, they aren't utilized until explicitly retrieved from the observations dictionary.

nilinykh, 2021-06-06:
Ah, I see! Thank you for the information.

> Please feel free to tag me for any future queries related to the EQA implementation.
Got it =)

closed 2021-06-06 by nilinykh