facebookresearch/habitat-lab #637: no semantic annotations

zhangfuyang, 2021-05-06:
Hi,

The setting I am using is: 

1.  v0.1.7 Habitat-sim ([link](https://github.com/facebookresearch/habitat-sim/tree/856d4b08c1a2632626bf0d205bf46471a99502b7)), installed by `conda install habitat-sim -c conda-forge -c aihabitat`
2.  v0.1.7 Habitat-lab ([link](https://github.com/facebookresearch/habitat-lab/tree/d6ed1c0a0e786f16f261de2beafe347f4186d0d8)), installed by `python setup.py develop --all`
3.  python3.6, ubuntu20

I am playing around with this demo tutorial https://aihabitat.org/docs/habitat-lab/habitat-lab-demo.html, but I can't get similar results shown on the page.

Particularly, for **Scene semantic annotations** part [link](https://aihabitat.org/docs/habitat-lab/habitat-lab-demo.html#scene-semantic-annotations), `scene.levels` is a blank list, shown below.
[image]

I guess this is the result why when I ran the **Actions and Sensors** part [link](https://aihabitat.org/docs/habitat-lab/habitat-lab-demo.html#actions-and-sensors), my result like this:
[image]

My config setting is here 
[image]

I wonder how to make the semantic sensor works? thanks!

aclegg3, 2021-05-06:
Hey @zhangfuyang, the semantic sensor requires an asset to come with semantic annotations. The skokloster-castle asset does not come with this information. You should try [MP3D](https://niessner.github.io/Matterport/), [Gibson](http://gibsonenv.stanford.edu/database/), or [Replica](https://github.com/facebookresearch/Replica-Dataset) assets instead.

zhangfuyang, 2021-05-07:
I see. Thanks

closed 2021-05-07 by zhangfuyang