facebookresearch/habitat-lab #641: Register a static camera in the env

gianscarpe, 2021-05-12:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim:  master
## ❓ Questions and Help

Hi all, thanks for the amazing work you're doing! I need to register a static camera in a fixed position in the scene, but I cannot really understand how to do it. Do you have any advice?
Many thanks!

dhruvbatra, 2021-05-12:
Have you looked at: https://aihabitat.org/docs/habitat-lab/view-transform-warp.html

closed 2021-05-20 by gianscarpe

dantes-ai, 2024-01-01:
> Have you looked at: https://aihabitat.org/docs/habitat-lab/view-transform-warp.html

The way described here uses the function  [get_observations_at](https://github.com/facebookresearch/habitat-lab/blob/e9850f22f3659bef97a7f7aeabbbabc6561f4fce/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py#L667) which seems kinda hacky... It moves agent 0 to the desired position then fetches the agent observations. This assumes that Agent 0 is has the required sensors. 

Is there a more direct way to probe an RGB image from anywhere in the scene, without utilizing an agent?