facebookresearch/habitat-lab #656: Project a single point from one view to the other

gianscarpe, 2021-06-15:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.1.7
Dear all,
I'm follwing the tutorial at https://aihabitat.org/docs/habitat-lab/view-transform-warp.html.
In particular, I have the camera of the agent and an observation obtained with `get_observation_at`, as explained in the tutorial.
I need to project one point from the agent's camera to the novel observation. To do so, I compute the intrinsics matrix using

FOV = 90
W = 512
Cx = Cy = W/2
Fx = Fy = (W/2) / np.tan(FOV / 2)
and construct the intrinsics matrix

GIven a point x,y in the original view, I extract the z value from the depth image and compute P as
P_cam0 = [(x - Cx) / Fx * z, (y - Cy) / Fy * z, z, 1]

Next, I build the transformation from cam0 to cam1 as explained in the tutorial and project P onto cam1:
P_cam1 = np.matmul(transformation, P_cam0)
Finally, I compute x_cam1, y_cam1 using the intrinsics matrix as build above
np.matmul(intrinsics_matrix, P_cam1) and divide by P_cam1[2]

Nontheless, the results I'm obtaining are not correct. Can you help me? Am I missing something?
Thanks :)

erikwijmans, 2021-06-17:
As that tutorial says, "negate depth as the camera looks along -Z", you need to negate the depth value.

rpartsey, 2022-09-05:
Hi @gianscarpe does the answer above resolve your issue? Thanks!

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity.

closed 2023-01-26 by rpartsey