facebookresearch/habitat-lab #659: Task sensors with get_observation_at

gianscarpe, 2021-06-24:
Hi there!
Is there a way to obtain also task sensors (like custom-defined sensors) when call "get_observation_at" and selecting a different viewpoints? The best solution I came up with is simply calling the get_observation function of a task-sensor with results from get_observation_at, but I cannot figured out how do that in practice.
Do you have any suggestions?
Thanks for your time :)

rpartsey, 2022-09-12:
Hi @gianscarpe!
The approach you described makes sense. 
What specifically are you struggling with?

In my opinion you can do it the same way it is implemented in the [HabitatSim.get_observations_at](https://github.com/facebookresearch/habitat-lab/blob/380ac0a7d8c4ead1532f109b15d329473212eae9/habitat/sims/habitat_simulator/habitat_simulator.py#L589):
```
sim: HabitatSim = ...

# set desired state 
sim.set_agent_state(position, rotation, reset_sensors=False)
sim_obs = sim.get_sensor_observations()
observations = sim._sensor_suite.sensors["sensor_uuid"].get_observation(sim_obs)

# reset state if needed
sim.set_agent_state(old_position, old_rotation, reset_sensors=False)
```

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity.

closed 2023-01-26 by rpartsey