facebookresearch/habitat-lab #670: X Error of failed request

Jiayuan-Gu, 2021-07-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

The habitat-sim is installed by conda following habitat-lab README. (I've also tried the version without `headless`)
```bash
conda install habitat-sim withbullet headless -c conda-forge -c aihabitat
```

Then, run the following minimal example (case 1).
```python
import habitat
import pygame

config = habitat.get_config("configs/tasks/pointnav.yaml")
with habitat.Env(config=config, dataset=None) as env:
    pygame.init()
    screen = pygame.display.set_mode([512, 512])
    print('Init pygame')
    obs = env.reset()
    print('Reset env')
    pygame.quit()
```

Or reverse the order of `pygame` and `env` (case 2)
```python
import habitat
import pygame

pygame.init()
print('Init pygame')
screen = pygame.display.set_mode([512, 512])
print('pygame screen')
config = habitat.get_config("configs/tasks/pointnav.yaml")
with habitat.Env(config=config, dataset=None) as env:
    obs = env.reset()
    print('Reset env')
    pygame.quit()
```

Interestingly, case 3 can work:
```
import habitat
import pygame

pygame.init()
print('Init pygame')
# screen = pygame.display.set_mode([512, 512])
# print('pygame screen')
config = habitat.get_config("configs/tasks/pointnav.yaml")
with habitat.Env(config=config, dataset=None) as env:
    obs = env.reset()
    print('Reset env')
    pygame.quit()
```

**I suspect that `pygame` and `habitat-sim` have conflict with something relevant to GL or X-server, as each library can work well seperately.**

Please note that without a minimal working example to reproduce the bug, we may not be able to help you.
[... 114 lines cut ...]
I0702 11:28:12.058820 11896 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0702 11:28:12.058859 11896 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0702 11:28:12.058884 11896 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0702 11:28:12.058898 11896 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0702 11:28:12.058912 11896 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0702 11:28:12.058946 11896 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0702 11:28:12.058974 11896 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0702 11:28:12.058990 11896 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0702 11:28:12.059003 11896 AssetAttributesManager.cpp:120] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0702 11:28:12.059026 11896 AssetAttributesManager.cpp:120] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0702 11:28:12.059046 11896 AssetAttributesManager.cpp:120] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0702 11:28:12.059054 11896 AssetAttributesManager.cpp:108] ::constructor : Built default primitive asset templates : 12
I0702 11:28:12.059630 11896 SceneDatasetAttributesManager.cpp:36] File (default) not found, so new default dataset attributes created and registered.
I0702 11:28:12.059635 11896 MetadataMediator.cpp:127] ::createSceneDataset : Dataset default successfully created.
I0702 11:28:12.059644 11896 AttributesManagerBase.h:365] <Physics Manager>::createFromJsonOrDefaultInternal : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I0702 11:28:12.059696 11896 PhysicsAttributesManager.cpp:26] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I0702 11:28:12.059702 11896 MetadataMediator.cpp:212] ::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
I0702 11:28:12.059707 11896 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
I0702 11:28:12.059713 11896 MetadataMediator.cpp:68] ::setSimulatorConfiguration : Set new simulator config for scene/stage : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb and dataset : default which is currently active dataset.
Platform::WindowlessEglContext: cannot make context current: EGL_BAD_ACCESS
Platform::WindowlessEglApplication::makeCurrent(): cannot make context current: EGL_BAD_ACCESS
Failed to make platform current
```

## Additional context

erikwijmans, 2021-07-06:
A headless version of the simulator likely doesn't work with pygame.  You can install the headed version via
```shell
conda remove habitat-sim headless
conda install habitat-sim -c conda-forge -c aihabitat
```

closed 2021-07-12 by dhruvbatra

guhur, 2021-11-16:
I'd like to re-open this issue (examples 1 and 2 fail with the same issue, not example 3). I am exactly on the same situation than @Jiayuan-Gu but I am not using `headless`. I have tested so far on master and last stable releases, while paying attention to cleaning previous installations. I am going to try on another machine.

ASzot, 2021-11-18:
@guhur is this still an issue on another machine? Can you also see if this works with the non-nightly build that Erik posted the installation instruction for above?

guhur, 2021-11-19:
Hi @ASzot yes it is working with latest version. I removed `habitat-lab` and `habitat-sim`, and re-do the install from source.