facebookresearch/habitat-lab #671: AttributeError: 'habitat_sim._ext.habitat_sim_bindings.CameraSensor' object has no attribute 'render_camera'

gtatiya, 2021-07-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7, commit: d6ed1c0a0e786f16f261de2beafe347f4186d0d8

Habitat-Sim: v0.1.7, commit: 856d4b08c1a2632626bf0d205bf46471a99502b7

## Docs and Tutorials
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Right now, I am going through that.

## ❓ Questions and Help

I am using the stable release v0.1.7 for Habitat-Sim and Habitat-Lab, and I am going through this tutorial: https://github.com/facebookresearch/habitat-lab/blob/v0.1.7/examples/tutorials/nb_python/Habitat_Interactive_Tasks.py. But, I get this error:

```
Traceback (most recent call last):
  File "examples/tutorials/nb_python/Habitat_Interactive_Tasks.py", line 753, in <module>
    sim, "rgb", crosshair_pos=[128, 190], max_distance=2.0
  File "examples/tutorials/nb_python/Habitat_Interactive_Tasks.py", line 706, in raycast
    render_camera = sim._sensors[sensor_name]._sensor_object.render_camera
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.CameraSensor' object has no attribute 'render_camera'
```

Could you please help me fix it?

erikwijmans, 2021-07-02:
Huh, looks like we tagged the wrong thing for 0.1.7.  bdd94e31eaa6f28036e92456b3ad3772a283b871 should have been what was tagged. @mathfac can we retag it?

mathfac, 2021-07-04:
@erikwijmans, the issue is that bdd94e31eaa6f28036e92456b3ad3772a283b871 doesn't have the version bump.

mathfac, 2021-07-10:
@gtatiya  are you using  bdd94e3 commit?  Otherwise I would advice to use that commit.

rpartsey, 2022-09-12:
Hi @gtatiya, have you managed to resolve the Issue?
Have you tried using commit suggested by Oleksandr (https://github.com/facebookresearch/habitat-lab/commit/bdd94e31eaa6f28036e92456b3ad3772a283b871)?

gtatiya, 2023-01-18:
> Hi @gtatiya, have you managed to resolve the Issue? Have you tried using commit suggested by Oleksandr ([bdd94e3](https://github.com/facebookresearch/habitat-lab/commit/bdd94e31eaa6f28036e92456b3ad3772a283b871))?

Yes, it is resolved, thank you!

closed 2023-01-18 by gtatiya