facebookresearch/habitat-lab #672: Expert Actions for Exploration

ugurbolat, 2021-07-07:
Hi Habitat devs!

I was wondering is there any dataset/environment (i.e., Replica, MP3D, etc.) that provides expert navigation actions for exploring the environment? I would like to experiment with building a semantic map, but it would be helpful if I can simplify the problem by disabling the action part for starters.

dhruvbatra, 2021-07-07:
CC: @vincentcartillier

vincentcartillier, 2021-07-07:
Hi @ugurbolat , this sounds very familiar. You should take a look at our Semantic MapNet project: 
- https://vincentcartillier.github.io/smnet.html
- https://github.com/vincentcartillier/Semantic-MapNet

In this project we build topdown semantic maps of an explored space (work done in MP3D and Replica). The exploration strategy here is assumed provided by an external policy or an oracle. In practice we've manually recorded those trajectories. And you can find those pre-recorded paths here: 
https://github.com/vincentcartillier/Semantic-MapNet/blob/main/data/paths.json

ugurbolat, 2021-07-07:
Hi @vincentcartillier, 

Thanks for the leads. This is exactly what I needed. Let me dig into your repo.

closed 2021-07-08 by dhruvbatra