facebookresearch/habitat-lab #676: I can't run the example/example.py

JiayunjieJYJ, 2021-07-08:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?
master
Habitat-Sim: vx.x.x or master?
master

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. cd habitat-lab
2. python example/example.py
Tail of the info in the terminal:

```

I0708 18:43:48.215437  4814 Sensor.cpp:69] Deconstructing Sensor
I0708 18:43:48.215448  4814 Sensor.cpp:69] Deconstructing Sensor
I0708 18:43:48.216286  4814 Renderer.cpp:71] Deconstructing Renderer
I0708 18:43:48.216351  4814 WindowlessContext.h:17] Deconstructing WindowlessContext
Traceback (most recent call last):
  File "examples/example.py", line 30, in <module>
    example()
  File "examples/example.py", line 19, in example
    observations = env.reset()  # noqa: F841
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/core/env.py", line 220, in reset
    self.reconfigure(self._config)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/core/env.py", line 303, in reconfigure
    self._sim.reconfigure(self._config.SIMULATOR)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 263, in reconfigure
    self._add_objs(ep_info)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 389, in _add_objs
    auto_sleep=self.habitat_config.get("AUTO_SLEEP", True),
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/obj_loaders.py", line 88, in load_articulated_objs
    ao.transformation = T
AttributeError: 'NoneType' object has no attribute 'transformation'
I0708 18:43:51.005597  4814 Simulator.cpp:54] Deconstructing Simulator
 
```
That's all about the error, please tell how to solve it. Thanks!

JiayunjieJYJ, 2021-07-08:
It is still a problem, I follow the step of this :https://github.com/facebookresearch/habitat-lab#installation, but I can't run example/example.py.

mathfac, 2021-07-08:
@JiayunjieJYJ , thank you for reporting this. Did you install Habitat Sim using Conda: `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`? Thank you.

JiayunjieJYJ, 2021-07-09:
> @JiayunjieJYJ , thank you for reporting this. Did you install Habitat Sim using Conda: `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`? Thank you.

JiayunjieJYJ, 2021-07-09:
> @JiayunjieJYJ , thank you for reporting this. Did you install Habitat Sim using Conda: `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`? Thank you.

Thank you for your reply. I install habitat-sim and habitat-lab on my virtual machine Ubuntu 18.04.5 on my laptop. The process was below :

```
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat

git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
cd habitat-lab
pip install -e .

git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
cd habitat-lab
pip install -r requirements.txt
python setup.py develop --all    # install habitat and habitat_baselines

conda install habitat-sim -c conda-forge -c aihabitat

cd habitat-lab
python example/example.py  # error occurs
```
After I run: **conda install habitat-sim withbullet headless -c conda-forge -c aihabitat** , I found the problem changed into this: 
```
(habitat) jiayunjie@jiayunjie-virtual-machine:~/robot_sim/habitat-lab$ python examples/example.py 
2021-07-09 00:07:19,063 Initializing dataset RearrangeDataset-v0
2021-07-09 00:07:22,019 initializing sim RearrangeSim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0709 00:07:22.026346  8287 ManagedFileBasedContainer.h:210] <Dataset>::convertFilenameToPassedExt : Filename : default changed to proposed scene_dataset_config.json filename : default.scene_dataset_config.json
I0709 00:07:22.026571  8287 AttributesManagerBase.h:365] <Dataset>::createFromJsonOrDefaultInternal : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0709 00:07:22.027642  8287 AssetAttributesManager.cpp:120] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0709 00:07:22.027801  8287 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0709 00:07:22.027890  8287 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0709 00:07:22.027946  8287 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0709 00:07:22.027977  8287 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0709 00:07:22.032249  8287 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0709 00:07:22.032472  8287 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0709 00:07:22.032546  8287 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0709 00:07:22.032583  8287 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0709 00:07:22.032613  8287 AssetAttributesManager.cpp:120] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0709 00:07:22.032660  8287 AssetAttributesManager.cpp:120] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0709 00:07:22.032702  8287 AssetAttributesManager.cpp:120] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0709 00:07:22.032718  8287 AssetAttributesManager.cpp:108] ::constructor : Built default primitive asset templates : 12
I0709 00:07:22.037420  8287 SceneDatasetAttributesManager.cpp:36] File (default) not found, so new default dataset attributes created and registered.
I0709 00:07:22.037514  8287 MetadataMediator.cpp:127] ::createSceneDataset : Dataset default successfully created.
I0709 00:07:22.037539  8287 AttributesManagerBase.h:365] <Physics Manager>::createFromJsonOrDefaultInternal : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I0709 00:07:22.041357  8287 PhysicsAttributesManager.cpp:26] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I0709 00:07:22.041527  8287 MetadataMediator.cpp:212] ::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
I0709 00:07:22.041534  8287 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
I0709 00:07:22.041553  8287 MetadataMediator.cpp:68] ::setSimulatorConfiguration : Set new simulator config for scene/stage : data/replica_cad/stages/Stage_v3_sc1_staging.glb and dataset : default which is currently active dataset.
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context

```

JiayunjieJYJ, 2021-07-09:
> @JiayunjieJYJ , thank you for reporting this. Did you install Habitat Sim using Conda: `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`? Thank you.

Could you please tell me how to solve the new problem above？

mathfac, 2021-07-10:
> on my virtual machine Ubuntu 18.04.5 on my laptop

Make sure that your GPU is accessible from Ubuntu.

You can set the environment variables for additional logging about why it couldn't find a device.

```
export MAGNUM_LOG=verbose 
export MAGNUM_GPU_VALIDATION=ON
```

Please look at https://github.com/facebookresearch/habitat-sim/issues/288 as it likely contains the answer.

closed 2021-07-12 by dhruvbatra

JiayunjieJYJ, 2021-09-14:
> > on my virtual machine Ubuntu 18.04.5 on my laptop
> 
> Make sure that your GPU is accessible from Ubuntu.
> 
> You can set the environment variables for additional logging about why it couldn't find a device.
> 
> ```
> export MAGNUM_LOG=verbose 
> export MAGNUM_GPU_VALIDATION=ON
> ```
> 
> Please look at [facebookresearch/habitat-sim#288](https://github.com/facebookresearch/habitat-sim/issues/288) as it likely contains the answer.

Thanks for your reply, there is a new issue which is my another question : [another issue](https://github.com/facebookresearch/habitat-lab/issues/719).

taylorlulu, 2023-10-26:
I met the same problem, after debugging I found that the problem was caused by the path of obj_path.
[image]
In the dictionary I found that there's no folder named configs in ycd folder. There're two version of ycb 1.1 and 1.2. The habitat-lab may use ycb1.2. After I renamed the folder ycb1.2 to ycb, the problem was solved