facebookresearch/habitat-lab #677: I can't run the example/example.py after install habitat-sim and habit-lab

JiayunjieJYJ, 2021-07-08:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:
1. cd habit-lab
2. python example/example.py
3. I can't run the example.py, tail of the info in terminal:
```

I0708 18:43:48.215437  4814 Sensor.cpp:69] Deconstructing Sensor
I0708 18:43:48.215448  4814 Sensor.cpp:69] Deconstructing Sensor
I0708 18:43:48.216286  4814 Renderer.cpp:71] Deconstructing Renderer
I0708 18:43:48.216351  4814 WindowlessContext.h:17] Deconstructing WindowlessContext
Traceback (most recent call last):
  File "examples/example.py", line 30, in <module>
    example()
  File "examples/example.py", line 19, in example
    observations = env.reset()  # noqa: F841
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/core/env.py", line 220, in reset
    self.reconfigure(self._config)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/core/env.py", line 303, in reconfigure
    self._sim.reconfigure(self._config.SIMULATOR)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 263, in reconfigure
    self._add_objs(ep_info)
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 389, in _add_objs
    auto_sleep=self.habitat_config.get("AUTO_SLEEP", True),
  File "/home/jiayunjie/robot_sim/habitat-lab/habitat/tasks/rearrange/obj_loaders.py", line 88, in load_articulated_objs
    ao.transformation = T
AttributeError: 'NoneType' object has no attribute 'transformation'
I0708 18:43:51.005597  4814 Simulator.cpp:54] Deconstructing Simulator
 

```

I dont know the reason about that, please tell me how to solve it.

erikwijmans, 2021-07-08:
Closing in favor or #676

closed 2021-07-08 by erikwijmans

JiayunjieJYJ, 2021-07-08:
> Closing in favor of #676
Can you please tell me how to solve it? #676 is also proposed by me, it is remain a problem, thanks!