facebookresearch/habitat-lab #678: Crash when running examples/tutorials/nhp_python/Habitat_Interactive_Tasks.py

jadkins99, 2021-07-08:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug
Get a segmentation fault (core dumped) error when trying to runthe Habitat_Interactive_Tasks.py tutorial code. 

## Steps to Reproduce

Steps to reproduce the behavior:

1.  python  examples/tutorials/nb_python/Habitat_Interactive_Tasks.py 

Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

[image]

## Expected behavior
Was expecting to run the Interactive Task Tutorial.

erikwijmans, 2021-07-08:
You need to download the required scene, run
```bash
python -m habitat_sim.utils.datasets_download --uids coda_scene --data-path ./data
```
to download that.

jadkins99, 2021-07-08:
Okay, thank you! After downloading the scene and running again, there is a IndexError: list out of range error. What is the cause of this?
[image]

erikwijmans, 2021-07-08:
Likely missing some additional assets, 
```shell
python -m habitat_sim.utils.datasets_download --uids ci_test_assets --data-path ./data
```
should get all the needed ones.

jadkins99, 2021-07-09:
Great! Thank you!

closed 2021-07-09 by jadkins99