facebookresearch/habitat-lab #681: Question on Home Assistant Benchmark Tasks/Actions

jadkins99, 2021-07-09:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I have a question about how to utilize the tasks and actions described in section 6 of the Habitat 2.0 paper. There was a discussion of TidyHouse, PrepareGroceries, SetTable as long range tasks studied and using a set of parametrized higher level skills such a Pick, Place, Open fridge door, etc.  
Currently, I can't find anything in the repo regarding these long-rank tasks and high-level actions. Can you please described where these tasks and actions are defined as well as how they can be utilized in a config file?

dhruvbatra, 2021-07-09:
The HAB tasks have not been yet been released. 

[image]

jadkins99, 2021-07-10:
Okay, thank you!

closed 2021-07-10 by jadkins99

jadkins99, 2021-08-10:
Any updates on when the HAB tasks and actions will be released?

Runjian-Chen, 2022-03-27:
Is there any update about the release of HAB tasks? Thank you!

ASzot, 2022-03-29:
Yes, they are almost ready for release. All the HAB tasks are public here: https://github.com/facebookresearch/habitat-lab/pull/837. We are working on cleaning up and documenting the code for release. 

I will follow up by the end of the week with a tutorial on how to run the HAB tasks.