facebookresearch/habitat-lab #682: gym distribution not found and is required by habitat

pbideau, 2021-07-12:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master
Habitat-Sim: master

## ❓ Questions and Help
It looks to me that gym is installed but somehow not found by habitat. Maybe someone has an idea how to solve this issue?
thank you very much in advance! 

```
(habitat) Pia@r2d2:~/Development/manipulation/habitat-lab$ sudo python setup.py develop --all
running develop
running egg_info
writing habitat.egg-info/PKG-INFO
writing dependency_links to habitat.egg-info/dependency_links.txt
writing requirements to habitat.egg-info/requires.txt
writing top-level names to habitat.egg-info/top_level.txt
reading manifest file 'habitat.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat.egg-info/SOURCES.txt'
running build_ext
Creating /usr/local/lib/python3.7/dist-packages/habitat.egg-link (link to .)
habitat 0.2.0 is already the active version in easy-install.pth

Installed /home/Pia/Development/manipulation/habitat-lab
Processing dependencies for habitat==0.2.0
Searching for gym>=0.17.3
Reading https://pypi.org/simple/gym/
Downloading https://files.pythonhosted.org/packages/1a/db/816fd52c0c196b6799e89d1f65b6c74fead2707cf7d447f3f354edfa7a44/gym-0.18.3.tar.gz#sha256=81a3e3fbf7fcf57c8cf98f7e22d1bdd5815f3824d9c148a7eb42420d3d642967
Best match: gym 0.18.3
Processing gym-0.18.3.tar.gz
Writing /tmp/easy_install-_08knjvh/gym-0.18.3/setup.cfg
Running gym-0.18.3/setup.py -q bdist_egg --dist-dir /tmp/easy_install-_08knjvh/gym-0.18.3/egg-dist-tmp-tizplzp8
removing '/usr/local/lib/python3.7/dist-packages/gym-0.2.0-py3.7.egg' (and everything under it)
creating /usr/local/lib/python3.7/dist-packages/gym-0.2.0-py3.7.egg
Extracting gym-0.2.0-py3.7.egg to /usr/local/lib/python3.7/dist-packages
gym 0.2.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python3.7/dist-packages/gym-0.2.0-py3.7.egg
error: The 'gym>=0.17.3' distribution was not found and is required by habitat

```

dhruvbatra, 2021-07-12:
CC: @mathfac

pbideau, 2021-07-13:
I am replying here to my own question :-)
The solution to this was to first uninstall the automatically installed gym 0.2.0 and then to manually install gym 0.18.3.
This way a gym version was found while running setup.py and not being installed again.

I actually do not quite understand why gym 0.18.3 was correctly identified as the best match (see terminal output), but gym 0.2.0 is getting installed.

Anyhow, this solved the problem for me and the example.py is running! Thank you :-)!

Skylion007, 2021-07-13:
@ASzot @mathfac  Looks like some potential OpenAI Gym API incompatibility. We might need to deal with this for the gym wrapper.

closed 2021-07-13 by pbideau