facebookresearch/habitat-lab #683: Question about Action Space

jadkins99, 2021-07-12:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I have a question about the action used in the  rearrangepick_replica_cad.yaml config used in example.py. What are the parameters in the list list of length 7 that is the key for arm_ac in the action dictionary? From reading the paper, I assumed the action space would be of the form  ([dx,dy,dz], grip_ac). I am confused by what are the 7 parameters in the list and where to go to read more about different action space configurations. 

Thanks for your help.

dhruvbatra, 2021-07-12:
CC: @ASzot

ASzot, 2021-07-13:
Hello, the 7 parameters are joint offset values for the robot arm (the Fetch robot arm has 7 arm joints). You can find the code [here](https://github.com/facebookresearch/habitat-lab/blob/9220a301c3b168f845150a3d2f20df8fc21f3842/habitat/tasks/rearrange/actions.py#L113) for setting the arm motor positions. You’re correct the paper used end-effector control rather than joint control for the action space. However, this code had a dependency on PyBullet therefore we didn’t include it in the initial Habitat 2.0 public code release. The release will be updated soon to include the end-effector control as well. In the future, we also plan on adding better documentation for the different action spaces. For now, you can find the different Habitat 2.0 rearrangement task action space definitions in [this file](https://github.com/facebookresearch/habitat-lab/blob/master/habitat/tasks/rearrange/actions.py).

jadkins99, 2021-07-15:
Great thank you!

closed 2021-07-15 by jadkins99