facebookresearch/habitat-lab #688: Issue When loading scene from scene. MeshTools::compile() invalid external buffer(s)

jadkins99, 2021-07-15:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hi, 
I receive the error below when attempting to run Habitat 2.0. I am using the rearrangepick_replica_cad.yaml config file and creating the config and env as in example.py with config = habitat.get_config(config_path=<path_to_config>) and env = habitat.env(config=config, dataset = dataset). 

[image]

I would appreciate any advice on what the cause of the error might be.

Best.

dhruvbatra, 2021-07-15:
CC: @mathfac @ASzot

erikwijmans, 2021-07-15:
That's likely due to some data corruption.  Can you re-run the data download, 
```bash
python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --data-path data --clean
python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --data-path data
```

jadkins99, 2021-07-16:
Okay thank you!

closed 2021-07-16 by jadkins99