facebookresearch/habitat-lab #691: [Errno 104] Connection reset by peer when loading multiple Gibson Scenes on multiple GPUs

ruddradev, 2021-07-20:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v1.5

Habitat-Sim: v1.5

## ❓ Questions and Help

Hello, 

I have been trying to find the source of this error for a while, and would be great if I get any direction. I know it is an earlier version of Habitat, but I don't have a lot of choice here since I am evaluating another paper's code, and rewriting it for a newer version of Habitat might be a bigger problem. 

I am trying to run the code that loads 25 scenes on 25 threads, over 2 A100 GPUs. The total GPU memory is 80 GB (40+40) with 25 CPU cores and 1 TB RAM. after the scenes load, I get the below error -

```
5707 File "main.py", line 695, in <module>

5708 main()

5709 File "main.py", line 83, in main

5710 envs = make_vec_envs(args)

5711 File "/builds/gitlab/rni-compute/embedded-system-and-robotics/cognitive-robotic-navigation/object-goal-navigation-sota/envs/__init__.py", line 7, in make_vec_envs

5712 envs = construct_envs(args)

5713 File "/builds/gitlab/rni-compute/embedded-system-and-robotics/cognitive-robotic-navigation/object-goal-navigation-sota/envs/habitat/__init__.py", line 145, in construct_envs

5714 zip(args_list, env_configs, range(args.num_processes))

5715 File "/builds/gitlab/rni-compute/embedded-system-and-robotics/cognitive-robotic-navigation/object-goal-navigation-sota/envs/habitat/utils/vector_env.py", line 139, in __init__

5716 read_fn() for read_fn in self._connection_read_fns

5717 File "/builds/gitlab/rni-compute/embedded-system-and-robotics/cognitive-robotic-navigation/object-goal-navigation-sota/envs/habitat/utils/vector_env.py", line 139, in <listcomp>

5718 read_fn() for read_fn in self._connection_read_fns

5719 File "/usr/lib/python3.6/multiprocessing/connection.py", line 250, in recv

5720 buf = self._recv_bytes()

5721 File "/usr/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes

5722 buf = self._recv(4)

5723 File "/usr/lib/python3.6/multiprocessing/connection.py", line 379, in _recv

5724 chunk = read(handle, remaining)

5725ConnectionResetError: [Errno 104] Connection reset by peer
```
I read in another issue that it might be due to lack of memory, but I tried it with much lower memory and lesser threads so that lesser scenes are loaded at once.

xuexidi, 2022-07-24:
@ruddradev 
Hello, did you solve this error？I meet the same error, and I have no idea how to fix it......please help me..... :(

ruddradev, 2022-07-24:
Hello @xuexidi. I don't remember exactly. But what I do remember, is that there is a particular docker image called cudagl that has the pre-requisites to run Habitat in a server. Also with that I installed the appropriate pytorch versions, and build habitat with --with-cuda flag on. Let me know if it works.