facebookresearch/habitat-lab #692: examples: interactive_play.py - segmentation fault

pbideau, 2021-07-20:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master
Habitat-Sim: stable (installed via conda)

## ❓ Questions and Help
I am trying to get the example code to run
if I run `python examples/interactive_play.py` I am getting a Segmentation fault error.

Fatal Python error: (pygame parachute) Segmentation Fault

_Current thread 0x00007f8cffe50340 (most recent call first):
  File "/home/Pia/anaconda3/envs/habitat-36/lib/python3.6/site-packages/habitat_sim/simulator.py", line 615 in draw_observation
  File "/home/Pia/anaconda3/envs/habitat-36/lib/python3.6/site-packages/habitat_sim/simulator.py", line 400 in get_sensor_observations
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 368 in step
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/tasks/rearrange/actions.py", line 68 in step
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/core/embodied_task.py", line 303 in step
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/tasks/rearrange/rearrange_task.py", line 48 in step
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/tasks/rearrange/rearrange_pick_task.py", line 144 in step
  File "/home/Pia/Development/manipulation/habitat-lab/habitat/core/env.py", line 267 in step
  File "examples/interactive_play.py", line 127 in step_env
  File "examples/interactive_play.py", line 248 in get_input_vel_ctlr
  File "examples/interactive_play.py", line 297 in play_env
  File "examples/interactive_play.py", line 376 in <module>
Aborted (core dumped)_

The error arises in line 292 of interactive_play.py, when the variable step_results is set. The error seems to be related to the renderer. The segmentation fault finally happens in line 615 of habitat-sim/simulator.py (last stable version).

Any idea how to fix this, or whether I am something is missing?
Thank you very much!

ASzot, 2021-07-30:
What system are you running this on? @mathfac, @aclegg3 any idea why this could be happening? I just verified the interactive play script works as expected on my MacBook. Can you try running the interactive play script on this branch: https://github.com/facebookresearch/habitat-lab/tree/hab2_motion_planning? It has some stability fixes which could make a difference.

Jaekyung-Cho, 2021-09-24:
I have same problem.

my system configuration is following
Ubuntu 18.04
python 3.6.13
os type 64-bit

Stefuniverse, 2021-11-13:
Hi,
we are using this repo for a project at university and one of my project partners found, that you can start the example by installing an earlier pygame version.

`pip install pygame==1.9.6`

I am not sure, if this could be a final solution for this issue, because I am not sure of the implications of using this earlier version but it can at least be a workaround for you.

guhur, 2021-11-19:
@Stefuniverse's  solution is working for me! Thanks

closed 2023-01-26 by rpartsey