facebookresearch/habitat-lab #693: navmesh for replica_cad

jadkins99, 2021-07-20:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
Hi I am running into an error when I attempt to load replica_cad. I get the following error about not having the navmesh file Stage_v3_sc1_staging.navmesh. I checked and it is true that there are no navmesh files under data/replica_cad/stages/. Only .glb files. How/where do I find the navmesh files for replica_cad?
I already attempted: 
1. python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --data-path data --clean
2. python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --data-path data

[image]

jturner65, 2021-07-20:
Hello

I'm sorry you're running into issues with this.  If you notice in your printout, the dataset being used is 'default', which is not what you want here.  

You need to specify the Scene Dataset Configuration file for ReplicaCAD (this file is named "replicaCAD.scene_dataset_config.json" and is in the root of the replicaCAD data directory in the dataset download).  You should add a field to your habitat-lab YAML config named SCENE_DATASET_CONFIG_FILE that holds the fully qualified filepath to the Scene Dataset config for the replicaCAD dataset.

jadkins99, 2021-07-23:
Okay, thanks so much for your help!

closed 2021-07-23 by jadkins99