facebookresearch/habitat-lab #694: Env seems not to read config file value after edit

sparisi, 2021-07-27:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master
Habitat-Sim: v0.2.0.2021.07.25 (nightly build) (master gives me errors)

## Docs and Tutorials
I checked this tutorial
https://github.com/facebookresearch/habitat-lab/blob/master/docs/pages/habitat-lab-demo.rst

## ❓ Questions and Help
What I need to do: I need to set `MAX_EPISODE_STEPS` and `SIMULATOR.RGB_SENSOR.WIDTH` (and height) from my python script, not from the yaml file. The yaml file has some placeholder values.

What I tried: following the tutorial, I either try to pass these variables as `opts` or `defrost` and `freeze` the config file after loading it, and before passing it to `NavRLEnv`.

What happens: `MAX_EPISODE_STEPS` and width / height are properly changed in the config file, but when I pass it to `NavRLEnv` the old value (yaml placeholder) is read.

Minimal example:

```
from habitat.datasets import make_dataset
from habitat_baselines.config.default import get_config
from habitat_baselines.common.environments import NavRLEnv

config_path = 'pointnav_apartment-0.yaml' # Custom yaml with ENVIRONMENT.MAX_EPISODE_STEPS set to 500

opts = ['BASE_TASK_CONFIG_PATH', config_path] # I need this or `get_config` will look for `pointnav_nomap.yaml` by default.
opts += ['ENVIRONMENT.MAX_EPISODE_STEPS', 100]

config = get_config(config_paths=config_path, opts=opts)

print(config.ENVIRONMENT.MAX_EPISODE_STEPS) # This outputs 100

dataset = make_dataset(id_dataset=config.TASK_CONFIG.DATASET.TYPE,
                       config=config.TASK_CONFIG.DATASET)

env = NavRLEnv(config=config, dataset=dataset)

print(env._env._max_episode_steps) # This outputs 500
```

The same happens if I try as in the tutorial
```
config.defrost()
config.ENVIRONMENT.MAX_EPISODE_STEPS = 100
config.freeze()

```

Not sure if I am doing something wrong or if it is a bug.

erikwijmans, 2021-07-27:
It should be `TASK_CONFIG.ENVIRONMENT.MAX_EPISODE_STEPS`

sparisi, 2021-07-27:
> It should be `TASK_CONFIG.ENVIRONMENT.MAX_EPISODE_STEPS`

It works, thanks! It also works for observation width and height, awesome :)

closed 2021-07-27 by sparisi