facebookresearch/habitat-lab #696: How to activate rendering in human mode?

sparisi, 2021-07-29:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master
Habitat-Sim: master

## Docs and Tutorials
I checked them, but I couldn't find anything for 'render' and 'human'.

## ❓ Questions and Help
I would like to render environments in human mode, to see something like Fig. 1 and 2 of [your paper](https://arxiv.org/pdf/2106.14405.pdf), and your [demo video](https://aihabitat.org/tutorial/2020/).
If I try to render with `mode='human` I get the error below, saying that `human sensor is not active`.

It seems that `human` mode is not available [here](https://github.com/facebookresearch/habitat-lab/blob/0e1d2afe04b5856e6b5b3fc561adcea414091be4/habitat/sims/habitat_simulator/habitat_simulator.py#L385), but [here](https://github.com/facebookresearch/habitat-lab/blob/9220a301c3b168f845150a3d2f20df8fc21f3842/habitat/core/vector_env.py#L548) there is something about it.

```
~/.local/lib/python3.8/site-packages/gym/core.py in render(self, mode, **kwargs)
    238 
    239     def render(self, mode='human', **kwargs):
--> 240         return self.env.render(mode, **kwargs)
    241 
    242     def close(self):

~/habitat-lab/habitat/core/env.py in render(self, mode)
    419 
    420     def render(self, mode: str = "rgb") -> np.ndarray:
--> 421         return self._env.render(mode)
    422 
    423     def close(self) -> None:

~/habitat-lab/habitat/core/env.py in render(self, mode)
    304 
    305     def render(self, mode="rgb") -> np.ndarray:
--> 306         return self._sim.render(mode)
    307 
    308     def close(self) -> None:

~/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py in render(self, mode)
    396 
    397         output = observations.get(mode)
--> 398         assert output is not None, "mode {} sensor is not active".format(mode)
    399         if not isinstance(output, np.ndarray):
    400             # If it is not a numpy array, it is a torch tensor

AssertionError: mode human sensor is not active

```

dhruvbatra, 2021-07-29:
CC: @ASzot

ASzot, 2021-07-30:
Hello, it sounds like what you want will be provided in our Gym adapter here: https://github.com/facebookresearch/habitat-lab/blob/268fa779ec6bcaa39c36d58faa60cd6bd815e340/habitat_baselines/utils/gym_adapter.py#L167. Currently, only the RGB format is supported. We haven't released the nicer-looking ReplicaCAD scenes yet from Figures 1 and 2 of the paper and supplementary video for a "human" mode. The current scenes are without the baked lighting. Once this data is available, there will be an option to render nicer-looking scenes with shadows and baked lighting.

sparisi, 2021-07-30:
Hi, thanks for your reply! 
I managed to render RGB images with a custom viewer (similar to your utility). I asked for the third-person view because this gives a better idea of what the agent is doing, and allows to check if it is learning something meaningful (regardless of the increasing trend of the reward). 
RGB images are ok-ish for that, so I guess I will just wait for future rendering features :) 
Thanks again!

ASzot, 2021-07-30:
Yes, the current gym wrapper I linked to should render in this 3rd person POV. Here is an example of a video I just generated of an agent trained to reach target end-effector locations in a minimal scene. The `render()` method from `HabGymWrapper` was used to generate this. 

https://user-images.githubusercontent.com/10453381/127584176-99aa3d4e-9d73-4b0f-a374-c8e02bf9cb6e.mp4

sparisi, 2021-07-31:
(I updated both sim and lab to the latest version).

The video you uploaded doesn't load/play (gray screen stuck at 0:00).

I tried using `HabGymWrapper`, but it gave me some errors. 
If I try it like in [this examples](https://github.com/facebookresearch/habitat-lab/blob/9fc385c3e09e4799fe471e22bcd3e774c8bd9c8e/test/test_gym_wrapper.py#L25), I do
```
# ... parse config
env_class = get_env_class(config.ENV_NAME)
env = make_env_fn(env_class=env_class, config=config)
env = HabGymWrapper(env)
```
and get
```
2021-07-31 07:13:07,430 Initializing task Nav-v0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/private/home/sparisi/mushroom-rl/mushroom_rl/environments/habitat_env.py", line 163, in __init__
    env = HabGymWrapper(env)
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat_baselines/utils/gym_adapter.py", line 113, in __init__
    "Cannot convert this action space, more than one action"
ValueError: Cannot convert this action space, more than one action
```

The task is a custom `pointnav` YAML that I modify to return top-down maps and to use Replica scenes (action and observation spaces are default).
Anyway, I saw that [the render function](https://github.com/facebookresearch/habitat-lab/blob/bc85d0961cef3b4a08bc9263869606109fb6ff0a/habitat_baselines/utils/gym_adapter.py#L213) simply calls `observations_to_image` after saving the last observation, so I did the same with my wrapper
```
    def render(self, mode='rgb_array'):
        if mode == "rgb_array":
            frame = observations_to_image(
                self.env._last_full_obs, self.env.unwrapped._env.get_metrics()
            )
        else:
            raise ValueError(f"Render mode {mode} not currently supported.")
```
`_last_full_obs` works like `_last_obs` in your adapter.

I then can use a viewer to render the frame, and this is what I get

[image]

Still a 1st person image, not 3rd person. Am I missing something? Should I add measurements somewhere to the YAML?
(Also, is there a viewer in sim or lab that I can use? Couldn't find one).

Thanks!

jadkins99, 2021-08-10:
I am having the same problem, did you solve it?

sparisi, 2021-08-11:
> I am having the same problem, did you solve it?

No sorry, I didn't. I am using the 1st person rendering at the moment.

ASzot, 2021-12-14:
This issue should now be resolved on `main`. 

Also, the `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!

rpartsey, 2022-09-12:
Hi @sparisi @jadkins99, have you managed to resolve the issue as @ASzot suggested?

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity.

closed 2023-01-26 by rpartsey