facebookresearch/habitat-lab #698: Question about ee_pos

jadkins99, 2021-07-30:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.0

Habitat-Sim: v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I have a question about the values in the observations dict for ee_pos. I assumed that the ee_pos would stay bounded within the limits of the ee_constraints as defined here in habitat-sim: https://github.com/facebookresearch/habitat-sim/blob/b630441a539b1a0b06d310bda09791ed3cf67ec1/habitat_sim/robots/fetch_robot.py#L21
However, when I have sampled from the action space using example.py I have observed ee_pos values outside of the those of the constraints. Are those ee_constraints not the boundary of possible ee_values?

ASzot, 2021-07-31:
Not necessarily, this only constrains the _target_ end-effector position in IK control: https://github.com/facebookresearch/habitat-lab/blob/bc85d0961cef3b4a08bc9263869606109fb6ff0a/habitat/tasks/rearrange/actions.py#L361. When controlling the arm via the 7DoF control, this does not apply. Thanks for pointing out how this is not clear, a future PR will better document this.

jadkins99, 2021-08-10:
Okay, thank you!
Where can we find the set of possible (x,y,z) end-effector positions that the agent can achieve in the environment?

ASzot, 2021-12-14:
The possible achievable EE positions are defined [here](https://github.com/facebookresearch/habitat-lab/blob/11f23e9ad89b9f018d22256b0e5f71d63fb0e975/habitat/tasks/rearrange/actions.py#L329). 

Also, the `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!

rpartsey, 2022-09-12:
Closing as resolved. @jadkins99 feel free to reopen if you still have questions.

closed 2022-09-12 by rpartsey