facebookresearch/habitat-lab #700: When running example.py: AttributeError: 'FetchRobot' object has no attribute 'base_transformation'

jadkins99, 2021-07-31:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1

Habitat-Sim: v0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

I am having issues with setting up habitat lab after trying to do a clean reinstall compute cluster. Previously this habiat-lab was working on there. Below are the exact steps I followed from installation to running example.py.

## Steps to Reproduce

Steps to reproduce the behavior:

I followed these step from the begging.
1. conda create -n habitat_2 python=3.6 cmake=3.14.0
2. conda activate habitat_2
3. git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
4. cd habitat-lab
5. pip install -r requirements.txt
6. python setup.py develop --all
7.  conda install habitat-sim withbullet headless -c conda-forge -c aihabitat
8. python examples/example.py
At this point, I got the follow output:
[image]
Therefore, in order to make sure I had the ReplicaCAD scene dataset and rearrange pick task dataset I ran the follow:
8. python -m habitat_sim.utils.datasets_download --uids replica_cad_dataset
9. python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --no-replace
Step 9 led to the same output as above so I re-ran with removing the --no-replace flag.
10. python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets
Finally, I ran example.py again and got the below error.
11.examples/example.py
[image]

Is there something wrong with the way I performed setup or are there any additional assets that are needed?

Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

## Additional context

dhruvbatra, 2021-08-02:
CC: @mathfac

heurainbow, 2021-08-02:
similar issue

hpf9017, 2021-12-01:
same problem

ASzot, 2021-12-14:
Hello, thanks for reaching out. Your issue should now be fixed on the most recent version. 

The `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!