facebookresearch/habitat-lab #701: How to take a video where the robot is captured?

AgentEXPL, 2021-08-02:
I want to make a video which records a moving robot/agent. The video is similar with the video in the [tutorial page](https://aihabitat.org/tutorial/2020/). At present, I know how to get the RGB image from the RGB sensor mounted on the agent. However, I don't know how to make the agent/robot be visible in the observation. In a word, the problem is as follows:

Whether the default agent (i.e., cylinder) can be set to be visible via any configuration? If yes, which configuration parameter is concerned? If no, how can I generate a video similar with the one in the [tutorial page](https://aihabitat.org/tutorial/2020/)? It will be of great help to me if any related examples/tutorials could be provided?

aclegg3, 2021-08-02:
Hey @AgentEXPL, the default agent does not have a visual shape, so there is nothing for the camera to render. You can easily attach any asset to the Agent's SceneNode including a primitive cylinder shape. Check out the last cell in [this Colab ](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb) for an example of attaching the body shape,  setting up a 3rd person camera, and rendering a video.

AgentEXPL, 2021-08-03:
Hi @aclegg3 , thanks a lot for your quick reply. Based on your answer, I finally obtained what I want. 

By the way, I found a spelling mistake in the [example](https://github.com/facebookresearch/habitat-sim/blob/master/examples/tutorials/nb_python/ECCV_2020_Interactivity.py) provided in habitat-sim, as follows. When I changed the value of the position of sensors, the observations did not change. The reason is that "color_sensor_3rd_person_spec.postition" rather than"xxx.position" was used in the example. Maybe this could be corrected.
-------------------------------------------------------------------------------------------------------------------------
        color_sensor_3rd_person_spec.**postition** = [
            0.0,
            settings["sensor_height"] + 0.2,
            0.2,
        ]

closed 2021-08-03 by AgentEXPL

AgentEXPL, 2021-08-04:
Hi, @aclegg3 . A strange thing occurs when I run the above example if I set "allow_sliding = False". The locobot is stuck at the corner of the wall from 00:10. However, it suddenly gets the banana which is originally located at the bathroom at 00:30. Does it means that the non-visible agent actually has moved into the bathroom during 00:10~00:30, even though the locobot (which is visible) is stuck outside the bathroom.

Another question is that. When the locobot is attached to the Agent's SceneNode, will the size of locobot affects the movement of the the agent? For example, a path may be navigable for the agent in one size (i.e., radius and height) but not be navigable for the locobot in another size. Or, the locobot and the agent are automatically in the same size?

Could you help me to figure this out?

https://user-images.githubusercontent.com/75942858/128104633-02780ac2-d0fc-4bbd-9027-78e8fda95ea4.mp4