facebookresearch/habitat-lab #704: LibGL error problem

nightkidwyxc, 2021-08-06:
## ❓ Questions and Help
Hi All, 

I am new to Habitat-lab, I was trying to install Habitat-lab& Habitat-sim on my Ubuntu 18.04. However, I got some errors show below after I run "python ./examples/example.py" :

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Platform::WindowlessGlxContext: cannot create core context, falling back to compatibility context: BadRequest (invalid request code or no such operation)
Platform::WindowlessGlxContext: cannot create context: BadRequest (invalid request code or no such operation)
WindowlessContext: Unable to create windowless context

I googled "libGL error: No matching fbConfigs or visuals found" and found some answers are related to Nivida graphic card isssues. Because my system's graphic card is AMD 5500M , so I want to check if Habitat is only supports Nivida or not. If not, anyone could provide any hints to solve this problem? I tried to search libgl.so under my /lib, and I could find it under my /x86-64-linux-gnu folder, and I check the link config, shows that it does linked. But the strange thing is no /mesa folder under /x86 folder. I tried to re-install mesa by "sudo apt install mesa-libGL-devel" etc, still nothing changed. I appreciate that anyone could solve this. Thank you so much!

erikwijmans, 2021-08-07:
We support non nvidia graphics cards for non-headless binaries, which is what you seem to be using based off `Platform::WindowlessGlxContext`.

I haven't dealt with this exact issue, but in general you want to use libgl from your graphics card vendor's drivers not mesa as mesa is generic software drivers and won't have hardware support.  One issue that happens from time to time is installing mesa after installing your graphic card's drivers and overwriting its libgl.  I'd try installing your graphics card's drivers

nightkidwyxc, 2021-08-17:
Sorry for delay reply. Thank you so for your suggestion. I  tried what you said and failed, but I found it might be because I am using virtual machine. Then I tried to build it on an Ubuntu machine, It works.

closed 2021-08-17 by nightkidwyxc