facebookresearch/habitat-lab #706: Order of goal location array

zxwang96, 2021-08-08:
Hi,

I'd like to know the order of the pose element in the `position` array. Is it (x,y,o) or (x,o,y)?
https://github.com/facebookresearch/habitat-lab/blob/cafea28a2d685e02b8ffe339d7dd556a0b1ea42d/examples/shortest_path_follower_example.py#L71

erikwijmans, 2021-08-08:
It's a position, so `(x, y, z)`.

zxwang96, 2021-08-09:
Hi, thanks for answering my question. But based on my experiment, I have to give the goal position in the format of (x, orientation, y) to the `get_next_action()` method of `ShortestPathFollower`. I don't see any translation of `goals[0].position` from (x,y,o) to (x,o,y), so I assume it should be (x, o, y)?

erikwijmans, 2021-08-09:
Neither `NavigationGoal.position` nor `get_agent_state(0).position` has an orientation/rotational component.  Both are simply a 3D position `(x, y, z)` where +Y is up.  The agent orientation is in `get_agent_state(0).rotation` while `NavigationGoal` has no orientation.

zxwang96, 2021-08-09:
I see, thank you so much!

closed 2021-08-09 by zxwang96