facebookresearch/habitat-lab #708: /opt/conda/envs/habitat/bin/python: Error while finding module specification for 'habitat_sim.utils.datasets_download' (AttributeError: module 'habitat_sim.utils' has no attribute '__path__')

zhaozj89, 2021-08-16:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: docker fairembodied/habitat:latest

Habitat-Sim: docker

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I ran this script https://github.com/facebookresearch/habitat-sim/blob/089f6a41474f5470ca10222197c23693eef3a001/datasets/HM3D.md to download dataset, but it reported this error. I have tried other dataset scripts, and the same error appeared.  I am using the lastest docker `fairembodied/habitat:latest`, and cannot figure out what is the problem. Thanks.

mathfac, 2021-08-16:
Hi Zhenjie,

The Docker image isn’t updated continuously and wasn’t built for Habitat
2.0.
You can rebuild it using next Docker file:

https://github.com/facebookresearch/habitat-lab/blob/master/Dockerfile

zhaozj89, 2021-08-17:
Thanks. It seems that `libhdf5-dev` needs to be added to the Docker file. It works now.

closed 2021-08-17 by zhaozj89