facebookresearch/habitat-lab #716: RL BaseLine Visual Encoder Output Size

RobertHalwass, 2021-09-01:
I have a question regarding the output dimension of the visual encoder. There seems to be a discrepancy in the paper.
In the paper (page 15, Appendix C) it says: 
`Given these modifications, each backbone produces a 1024x4x4 feature map. We then convert this to a 128x4x4 feature map with a 3x3-Conv.`

But on the next page (page 16) it says:

`Our navigation policy takes the 64x4x4 feature map from the visual encoder, flattens it, and then converts the 2048-d vector to the same size as the hidden size via a fully-connected layer.`

Why was the number of channels halved from 128 to 64? I couldn't find an explanation in the code either. When I execute the code, it produces an output of 128x4x4, as was written on page 15.

dhruvbatra, 2021-09-01:
This repository is associated with a number of papers. I can't tell which paper you are talking about. Please email the first author directly (this isn't quite a question for github).

RobertHalwass, 2021-09-01:
Oh sorry, I should have mentioned the title of the paper. 
My comment was in regards to "DD-PPO: Learning Near-Perfect PointGoal Navigators from 2.5 Billion Frames" (https://arxiv.org/abs/1911.00357), the source for the RL-baseline. 
Since the author of the paper (Erik Wijmans) answered a question of mine here before and the question is in direct regard to a part of this repository, I felt it was appropriate to ask the question here.

erikwijmans, 2021-09-01:
The second one is a typo, it's `128x4x4`

RobertHalwass, 2021-09-01:
I see, thanks.

closed 2021-09-01 by RobertHalwass