facebookresearch/habitat-lab #718: I can't run the quick start example

JiayunjieJYJ, 2021-09-10:
## ❓ Questions and Help
I can run the QuickStart example.
There are so many dataset url for **gibson** dataset, as shown in the Readme.md: https://github.com/facebookresearch/habitat-lab#Gibson  

I can't download the right **gibson** dataset so I can't run the quickstart, could you please tell my the exact download url of the  **gibson** dataset?  

[image]

I need the .glb file like this:  
[image]

Thanks a lot !!!

JiayunjieJYJ, 2021-09-10:
I can't run the quick start example

JiayunjieJYJ, 2021-09-10:
Could you please tell me the exact download url of the gibson dataset which is 1.5GB and contains the '.glb' files?

dhruvbatra, 2021-09-10:
https://github.com/facebookresearch/habitat-sim/blob/master/DATASETS.md

closed 2021-09-10 by dhruvbatra