facebookresearch/habitat-lab #719: Can Habitat handle multi-agent scenarios?

JiayunjieJYJ, 2021-09-13:
## My purpose:
My purpose is to deal with the scene where multiple robots work together (such as multi-agent navigation, moving objects together, etc.), which requires:
1. Each robot can **see** other robots and other dynamic obstacles (such as humans).
2. Each robot can **detect** whether it collides with other robots.
3. Robots can **communicate** with each other (information can be their own observation or position, etc.).
4. Each robot can execute its own strategy **simultaneously in a distributed manner.**

## Thanks a lot!!!

JiayunjieJYJ, 2021-09-13:
Can Habitat add multiple robots in a single common environment?

erikwijmans, 2021-09-14:
For questions 1 and 2, this is something that could be supported without a large undertaking, we support adding render and collision assets to agents as well as creating multiple agents, but the APIs to do this easily don't exist.

For the policy questions, questions 3 and 4, this would require a much larger undertaking as the training/policy code is quite single agent/single policy focused.

While multi-agent is something we are interested in, full-fledged support isn't on any current roadmap that I am aware of it.

JiayunjieJYJ, 2021-09-14:
> For questions 1 and 2, this is something that could be supported without a large undertaking, we support adding render and collision assets to agents as well as creating multiple agents, but the [...]

@erikwijmans 
Thanks for your reply!   
So you mean that all the above requirements can be implemented, but there is no API directly available to call, that is, I need to do more work based on the existing API (but it can be implemented in the end)?

Skylion007, 2021-09-14:
There actually is some support on the Habitat-Sim side for multiple agents, but things like visualizing the other agents and handling collisions between agent is not compete there.

JiayunjieJYJ, 2021-09-15:
> There actually is some support on the Habitat-Sim side for multiple agents, but things like visualizing the other agents and handling collisions between agent is not compete there.

@Skylion007 Thanks! But as @erikwijmans said, '_For questions 1 and 2, this is something that could be supported without a large undertaking, we support adding render and collision assets to agents as well as creating multiple agents, but the APIs to do this easily don't exist.'_     

Is it possible if I implement these functions using the existing APIs? Or these function can't be supported at all?

JiayunjieJYJ, 2021-09-15:
@erikwijmans Another question is that can I load URDF file (the robot description file widely used in ROS) and control it in Habitat ?

Thanks a lot for your patience!

aclegg3, 2021-09-15:
> @erikwijmans Another question is that can I load URDF file (the robot description file widely used in ROS) and control it in Habitat ?

Yes, Habitat does support URDF. See python tests starting [here](https://github.com/facebookresearch/habitat-sim/blob/e51d955eae7aea0ed6406e54d2cf071913c9e2a1/tests/test_physics.py#L638) for example code. Also see [ManagedBulletArticulatedObject](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletArticulatedObject.html) API reference.

JiayunjieJYJ, 2021-09-16:
> Yes, Habitat does support URDF. See python tests starting [here](https://github.com/facebookresearch/habitat-sim/blob/e51d955eae7aea0ed6406e54d2cf071913c9e2a1/tests/test_physics.py#L638) for example [...]

Thanks a lot for your kindly reply!!! So for my requirement, the question turns to be this: by using the existing APIs, can I handle the questions I mentioned above:

> ## My purpose:
> My purpose is to deal with the scene where multiple robots work together (such as multi-agent navigation, moving objects together, etc.), which requires:
> 
> 1. Each robot can **see** other robots and other dynamic obstacles (such as randomly walking humans).
> 2. Each robot can **detect** whether it collides with other robots.
> 3. Robots can **communicate** with each other (the communication message can be their own observations or positions, etc.).
> 4. Each robot can execute its own policy  **simultaneously in a distributed manner.**

Or these functions can't be implemented at all now?  

That's very important for me because this will determine whether I need to study habitat in depth.

aclegg3, 2021-09-16:
Your points 1-3 could be implemented with Habitat-sim using existing APIs.

Regarding (4): This sounds like a learning API for simultaneous inference and training of multiple distinct policies within the same environment. While I believe you could implement such a system, it is not currently implemented within Habitat-lab and you would need to design it.

JiayunjieJYJ, 2021-09-18:
> Your points 1-3 could be implemented with Habitat-sim using existing APIs.
>
> Regarding (4): This sounds like a learning API for simultaneous inference and training of multiple distinct policies [...]

Thanks!! Nice to hear that!

closed 2021-09-18 by JiayunjieJYJ

huangyize-me, 2025-04-21:
> Thanks!! Nice to hear that!

Hello! Have you successfully built a multi-robot environment in Habitat? Could you give me some guidance? I have the same need to implement multi agent in Habitat? Thank you so much.