facebookresearch/habitat-lab #723: ImportError: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found

shuishida, 2021-09-16:
Hello. I have installed the most recent version of habitat-lab from the master branch, and installed habitat-sim via the headless option 
`conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`

I am trying to run the DD-PPO training script indicated here: https://github.com/facebookresearch/habitat-challenge#pointnavobjectnav-baselines-and-dd-ppo-training-starter-code

When I tried running the [DD-PPO script](https://github.com/facebookresearch/habitat-lab/blob/master/habitat_baselines/rl/ddppo/single_node.sh) by
`sh habitat_baselines/rl/ddppo/single_node.sh` from the habitat-lab directory, it gave me an error 
`ImportError: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found`

I suspected an gcc version mismatch, and tried compiling `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat` with gcc 4.8.5, 5.4.0 and 9.3.0 but none of them worked. 

Would appreciate any pointers!

Please find attached the error message screen shot.
[image]

erikwijmans, 2021-09-17:
Conda ships its own libstdc.  Doing `ls ${CONDA_PREFIX}/lib | grep stdc` should show it. That should also get picked up automatically, but you can add `export LD_LIBRARY_PATH=${CONDA_PREFIX}/lib:${LD_LIBRARY_PATH}` to help things out.

shuishida, 2021-09-17:
Wow, you are amazing!! Adding the latter solved it! Thank you so much!!

closed 2021-09-17 by shuishida