facebookresearch/habitat-lab #735: How to train an agent that navigates to a set of possible goal locations

yhymason, 2021-09-27:
Hi there, I am planning to train my own embodied agent using Habitat-lab and Habitat-sim. However, I have a few questions before I start:
1, Are there any good examples/tutorials for agent-training using the Habitat framework? 
2, How can I formulate the training data and configs so that the agent can learn to navigate to one of the possible goal locations in a scene instead of just one fixed goal location? 

Thanks in advance for all answers!

dhruvbatra, 2021-09-27:
https://aihabitat.org/tutorial/2020/

closed 2021-09-27 by dhruvbatra

yhymason, 2021-09-27:
@dhruvbatra thanks for replying, I have seen this tutorial before but it seems not to cover setting up the training process directly. Also, I don't think it covers my second question from above. Can you provide additional guidance?

erikwijmans, 2021-09-28:
The habitat_baselines folder contains examples of training with habitat. https://github.com/facebookresearch/habitat-lab/tree/master/habitat_baselines/rl/ppo is likely a good place to start.

For the second ask, that is entirely possible to do, but not without code changes as we currently don't have support for pointnav with more than 1 goal implemented.

yhymason, 2021-09-28:
@erikwijmans thanks for the additional suggestions! And yes!, I am also thinking about refactoring/inheriting some of the existing classes under habitat-lab to make the training work with multiple goal locations. However, it is not very clear to me which classes I should make changes to. So far I can only think of the embodied task class, but what else am I missing?

erikwijmans, 2021-09-30:
You'll need to make changes to the DistanceToGoal measure and the PointGoal sensor (https://github.com/facebookresearch/habitat-lab/blob/master/habitat/tasks/nav/nav.py). You'll need to make changes to the training code if the number of goals isn't constant/can't be padded to be constant.

yhymason, 2021-10-21:
@erikwijmans hi Erik, sorry for following up after so many days, as you might remember, I was working on generating point clouds using depth images in habitat sim. Now I finally have time to revisit this side of work. As you suggested, I looked into nav.py. I see that DistanceToGoal measure already has the following logic:
`    def update_metric(
        self, episode: NavigationEpisode, *args: Any, **kwargs: Any
    ):
        current_position = self._sim.get_agent_state().position

        if self._previous_position is None or not np.allclose(
            self._previous_position, current_position, atol=1e-4
        ):
            if self._config.DISTANCE_TO == "POINT":
                distance_to_target = self._sim.geodesic_distance(
                    current_position,
                    [goal.position for goal in episode.goals],
                    episode,
                )`
It looks like this Measure class already can handle multiple goals, does it mean I only need to change the logic in PointGoal sensor to make things work? 
Also, can you show me how to define multiple goal locations in each training episode?

erikwijmans, 2021-11-02:
As shown in the tutorial, there is a `config.TOTAL_NUM_STEPS` field that determines the total number of steps of experience to train for. Setting that to something very large, i.e. `1e8`, will cause it it train more.

yhymason, 2021-11-02:
@erikwijmans Thanks Erik, can I confirm how `config.TOTAL_NUM_STEPS` relates to training over all episodes? say I have scene A with a total of 80k episodes, should I set `config.TOTAL_NUM_STEPS` to at least 80k * 500 to make the training process cover all possible episodes?

yhymason, 2021-11-03:
@erikwijmans I looked further and found inside config:
```
TASK_CONFIG:
  DATASET:
    CONTENT_SCENES: ['*']
    DATA_PATH: data/datasets/pointnav/habitat-test-scenes/v1/{split}/{split}.json.gz
    SCENES_DIR: data/scene_datasets
    SPLIT: train
    TYPE: PointNav-v1
  ENVIRONMENT:
    ITERATOR_OPTIONS:
      CYCLE: True
      GROUP_BY_SCENE: True
      MAX_SCENE_REPEAT_EPISODES: -1
      MAX_SCENE_REPEAT_STEPS: 10000
      NUM_EPISODE_SAMPLE: -1
      SHUFFLE: True
      STEP_REPETITION_RANGE: 0.2
    MAX_EPISODE_SECONDS: 10000000
    MAX_EPISODE_STEPS: 500
```
Can you explain the relationship between `config.TOTAL_NUM_STEPS` and `MAX_EPISODE_STEPS: 500`, and how many steps will each episode run?

erikwijmans, 2021-11-03:
In general, we don't track the number of episodes scene during training. We haven't found number of episodes trained on to be a meaningful way to measure the amount of training since episodes can have a variable number of steps in them. 

But yes, if you set TOTAL_NUM_STEPS to something greater than the number of episodes times the maximum episode length you will see all episodes at least once.