facebookresearch/habitat-lab #736: How to pass STOP to the rearrange task

sparisi, 2021-09-27:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim: master

## ❓ Questions and Help
According to the paper, the STOP action is needed to end the episode and get the success reward in the rearrange task. However, the task action space accepts only ARM_ACTION with kwargs 'arm_action' and 'grip_action'. I tried passing STOP in many ways, but I always get an error.
How do I tell the agent to STOP?

ASzot, 2021-12-14:
In the Habitat 2.0 paper, each skill had a custom termination condition. For example, for the Pick skill this was based on if the end-effector was at the resting position with an object grasped. However, the stop discrete action can easily be added to the action space through the `POSSIBLE_ACTIONS` key in the task config ([example](https://github.com/facebookresearch/habitat-lab/blob/hab_suite/configs/tasks/rearrange/pick.yaml#L110)). 

The `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!

sparisi, 2021-12-14:
@ASzot 
Thanks for your reply, I will check the tutorial!
Just to double check: the task does not require `STOP` to terminate, right? 
The YAMLs I see all only have `ARM_ACTION` in `POSSIBLE_ACTIONS`, and this does not allow to pass `STOP`. Also, the [nav task](https://github.com/facebookresearch/habitat-lab/blob/hab_suite/habitat/tasks/nav/nav.py) explicitly checks for `STOP`, while [the rearrange task](https://github.com/facebookresearch/habitat-lab/blob/hab_suite/habitat/tasks/rearrange/rearrange_task.py) does not.

Yet, the paper says that `STOP` is required.
> Success: The agent is within 0.3 meters of the goal, 0.5 radians of the target angle, and has called
the stop action at the current time step.

Am I missing something in the code, or does the paper has just some "wrong wording"? 

Thanks!

sparisi, 2021-12-14:
@ASzot 
Thanks for your reply, I will check the tutorial!
I think I also got confused because the paper says that only 'Navigate' --and not any 'Rearrange'-- needs `STOP` to terminate.

closed 2021-12-14 by sparisi