facebookresearch/habitat-lab #737: Habitat 2.0 Base Velocity Control Errors Out in example code

mihdalal, 2021-09-28:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1 or master?

Habitat-Sim: v0.2.1 or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

I tried modifying the rearrange pick task to involve moving the base as well. When sampling random actions according to the action space, I get a dictionary error. I have included the modified parts of rearrange_pick_replica_cad.yaml as well as the command and error below. 

## Steps to Reproduce
Modify rearrange_pick_replica_cad.yaml with the following:
```
ACTIONS:
        ARM_ACTION:
            TYPE: "ArmAction"
            ARM_CONTROLLER: "ArmRelPosAction"
            GRIP_CONTROLLER: "MagicGraspAction"
            ARM_JOINT_DIMENSIONALITY: 7
            GRASP_THRESH_DIST: 0.15
            DISABLE_GRIP: False
            DELTA_POS_LIMIT: 0.0125
        BASE_VELOCITY:
            TYPE: "BaseVelAction"
            LIN_SPEED: 12.0
            ANG_SPEED: 12.0
            ALLOW_DYN_SLIDE: True
            END_ON_STOP: False
            ALLOW_BACK: True
            MIN_ABS_LIN_SPEED: 1.0
            MIN_ABS_ANG_SPEED: 1.0
    POSSIBLE_ACTIONS:
        - BASE_VELOCITY
```
Then run `python examples/example.py`

Output: 
```
Traceback (most recent call last):
  File "examples/example.py", line 30, in <module>
    example()
  File "examples/example.py", line 24, in example
    observations = env.step(env.action_space.sample())  # noqa: F841
  File "/home/mdalal/research/mobile_manip/habitat-lab/habitat/core/env.py", line 267, in step
    action=action, episode=self.current_episode
  File "/home/mdalal/research/mobile_manip/habitat-lab/habitat/tasks/rearrange/rearrange_pick_task.py", line 133, in step
    obs = super().step(action=action, episode=episode)
  File "/home/mdalal/research/mobile_manip/habitat-lab/habitat/tasks/rearrange/rearrange_task.py", line 68, in step
    obs = super().step(action=action, episode=episode)
  File "/home/mdalal/research/mobile_manip/habitat-lab/habitat/core/embodied_task.py", line 303, in step
    observations = task_action.step(**action["action_args"], task=self)
TypeError: step() argument after ** must be a mapping, not numpy.ndarray
```
## Expected behavior

No error, 200 steps in the environment are completed.

ASzot, 2021-12-14:
Hello, thanks for reaching out. Your issue should now be fixed on the most recent version. 

The `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!

mihdalal, 2021-12-15:
I am still getting the same error even after having updated to the most recent branch. Can you try running the same test and let me know what you observe?

ASzot, 2021-12-15:
This works fine for me. The example now loads the config file `configs/tasks/rearrange/pick.yaml`. I modified this config with the action space you describe above and it runs without an issue. Can you send the exact config file you are using? Can you also confirm you are on the `hab_suite` branch of Habitat Lab by checking that `example/example.py` uses the `configs/tasks/rearrange/pick.yaml` config file?

mihdalal, 2021-12-15:
Okay for this one: it works on `hab_suite` and does not work on `main` just yet.

ASzot, 2021-12-15:
Yeah, it is best to only use `hab_suite` until things are fully merged into `main`.

rpartsey, 2022-09-12:
@mihdalal Feel free to re-open the issue, if you still have questions.

closed 2022-09-12 by rpartsey